uos/sick_tim #27: Sick Tim571 ROS interface problem, device not detected

sacuar, 2015-05-23:
I was trying to connect a Sick Tim571 to ubuntu 14.04 LTS 64 bit. The device connected and worked properly in windows/SOPAS ET. But I failed to connect it in ROS! I pinged the device successfully and the tcp/ip worked well in MATLAB too. I modified the time limit in the launch file and the IP address according to my network. But still , no success! I tried to find the answer in ROS wiki and in the ROS discussion page. But got no answer/reply.

Would be very kind of you, if you have some solution to the problem.

ping status ..

 netcat -vt 10.120.6.9 2112

Connection to 10.120.6.9 2112 port [tcp/*] succeeded!

Below is the log file.

1432422685.497675320  Node Startup
1432422686.468419398 WARN [/tmp/buildd/ros-indigo-kdl-parser-1.11.6-0trusty-20150326-1841/src/kdl_parser.cpp:174(treeFromUrdfModel) [topics: /rosout] The root link laser_mount_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
1432422686.692652176 ERROR [/home/sacuar/catkin_ws/src/sick_tim-indigo/src/sick_tim_common_usb.cpp:349(SickTimCommonUsb::init_device) [topics: /rosout, /diagnostics, /sick_tim551_2050001/parameter_descriptions, /sick_tim551_2050001/parameter_updates, /scan] No SICK TiM3xx devices connected!
1432422686.692888940 FATAL [/home/sacuar/catkin_ws/src/sick_tim-indigo/src/sick_tim_common.cpp:98(SickTimCommon::init) [topics: /rosout, /diagnostics, /sick_tim551_2050001/parameter_descriptions, /sick_tim551_2050001/parameter_updates, /scan] Failed to init device: 1
1432422687.693543826 ERROR [/home/sacuar/catkin_ws/src/sick_tim-indigo/src/sick_tim_common_usb.cpp:266(SickTimCommonUsb::sendSOPASCommand) [topics: /rosout, /diagnostics, /sick_tim551_2050001/parameter_descriptions, /sick_tim551_2050001/parameter_updates, /scan] LIBUSB - device not open
1432422687.854591294 ERROR [/home/sacuar/catkin_ws/src/sick_tim-indigo/src/sick_tim_common_usb.cpp:349(SickTimCommonUsb::init_device) [topics: /rosout, /diagnostics, /sick_tim551_2050001/parameter_descriptions, /sick_tim551_2050001/parameter_updates, /scan] No SICK TiM3xx devices connected!
1432422687.855758772 FATAL [/home/sacuar/catkin_ws/src/sick_tim-indigo/src/sick_tim_common.cpp:98(SickTimCommon::init) [topics: /rosout, /diagnostics, /sick_tim551_2050001/parameter_descriptions, /sick_tim551_2050001/parameter_updates, /scan] Failed to init device: 1
1432422688.856983239 ERROR [/home/sacuar/catkin_ws/src/sick_tim-indigo/src/sick_tim_common_usb.cpp:266(SickTimCommonUsb::sendSOPASCommand) [topics: /rosout, /diagnostics, /sick_tim551_2050001/parameter_descriptions, /sick_tim551_2050001/parameter_updates, /scan] LIBUSB - device not open
1432422689.008165456 ERROR [/home/sacuar/catkin_ws/src/sick_tim-indigo/src/sick_tim_common_usb.cpp:349(SickTimCommonUsb::init_device) [topics: /rosout, /diagnostics, /sick_tim551_2050001/parameter_descriptions, /sick_tim551_2050001/parameter_updates, /scan] No SICK TiM3xx devices connected!
1432422689.008960766 FATAL [/home/sacuar/catkin_ws/src/sick_tim-indigo/src/sick_tim_common.cpp:98(SickTimCommon::init) [topics: /rosout, /diagnostics, /sick_tim551_2050001/parameter_descriptions, /sick_tim551_2050001/parameter_updates, /scan] Failed to init device: 1
1432422690.010001793 ERROR [/home/sacuar/catkin_ws/src/sick_tim-indigo/src/sick_tim_common_usb.cpp:266(SickTimCommonUsb::sendSOPASCommand) [topics: /rosout, /diagnostics, /sick_tim551_2050001/parameter_descriptions, /sick_tim551_2050001/parameter_updates, /scan] LIBUSB - device not open
1432422690.194069949 ERROR [/home/sacuar/catkin_ws/src/sick_tim-indigo/src/sick_tim_common_usb.cpp:349(SickTimCommonUsb::init_device) [topics: /rosout, /diagnostics, /sick_tim551_2050001/parameter_descriptions, /sick_tim551_2050001/parameter_updates, /scan] No SICK TiM3xx devices connected!
1432422690.195063025 FATAL [/home/sacuar/catkin_ws/src/sick_tim-indigo/src/sick_tim_common.cpp:98(SickTimCommon::init) [topics: /rosout, /diagnostics, /sick_tim551_2050001/parameter_descriptions, /sick_tim551_2050001/parameter_updates, /scan] Failed to init device: 1
1432422691.196088846 ERROR [/home/sacuar/catkin_ws/src/sick_tim-indigo/src/sick_tim_common_usb.cpp:266(SickTimCommonUsb::sendSOPASCommand) [topics: /rosout, /diagnostics, /sick_tim551_2050001/parameter_descriptions, /sick_tim551_2050001/parameter_updates, /scan] LIBUSB - device not open
1432422691.494045635 ERROR [/home/sacuar/catkin_ws/src/sick_tim-indigo/src/sick_tim_common_usb.cpp:349(SickTimCommonUsb::init_device) [topics: /rosout, /diagnostics, /sick_tim551_2050001/parameter_descriptions, /sick_tim551_2050001/parameter_updates, /scan] No SICK TiM3xx devices connected!

sacuar, 2015-05-23:
$ roslaunch sick_tim sick_tim551_2050001.launch
... logging to /home/sacuar/.ros/log/581708f8-01a7-11e5-9694-a0b3ccbe9eee/roslaunch-sacuar-Presario-CQ43-Notebook-PC-20151.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:33801/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.10

NODES
  /
    robot_state_publisher (robot_state_publisher/state_publisher)
    sick_tim551_2050001 (sick_tim/sick_tim551_2050001)

auto-starting new master
process[master]: started with pid [20166]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 581708f8-01a7-11e5-9694-a0b3ccbe9eee
process[rosout-1]: started with pid [20179]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [20182]
process[sick_tim551_2050001-3]: started with pid [20183]
[ERROR] [1432425401.703083624]: No SICK TiM3xx devices connected!
[FATAL] [1432425401.703896199]: Failed to init device: 1
[ERROR] [1432425402.704331211]: LIBUSB - device not open

SOPAS - Error stopping streaming scan data!
sick_tim driver exiting.
[ERROR] [1432425402.860435036]: No SICK TiM3xx devices connected!
[FATAL] [1432425402.860723692]: Failed to init device: 1
[ERROR] [1432425403.861084161]: LIBUSB - device not open

SOPAS - Error stopping streaming scan data!

mintar, 2015-05-25:
**sick_tim551_2050001.launch**

    <?xml version="1.0"?>
    <launch>
      <param name="robot_description" command="$(find xacro)/xacro.py '$(find sick_tim)/urdf/example.urdf.xacro'" />
      <node name="robot_state_publisher" pkg="robot_state_publisher" type="state_publisher" />
    
      <node name="sick_tim551_2050001" pkg="sick_tim" type="sick_tim551_2050001" respawn="false" output="screen">
        
          
    
          
      </node>
    </launch>

mintar, 2015-05-25:
The last three parameters are still commented out, so the driver is ignoring IP etc. and trying USB instead (like the error message says). Remove the `` around them, and it should work.

Also, 500 seconds timeout seems a bit long.

sacuar, 2015-05-26:
Thanks mintar! I have been doing that mistake so long!!! Just did not see those were commented!!!! it works now!I am getting replies from the scanner. I can tell more after visualizing in rviz. That node works with the new Tim571 :) Great!

closed 2015-05-26 by sacuar

TerryGP, 2015-12-09:
Hello, I have same problem here. 
I was trying to connect a Sick Tim 561 to ubuntu 14.04 LTS 64 bit. The device connected and worked properly in windows/SOPAS ET. But I failed to connect it in ROS. I  can ping the device successfully and I modified the IP address according to my network. But still not work... 

I also publish this issue in the ROS answer, but there is no answer now.
Would be very kind of you, if you have some solution to the problem.
The device used 0.333 degree resolution, Ethernet connection, scanning frequency 15 Hz.

As below is the ping status :

terry@terry-All-Series:~$ nc -v 192.168.1.144 2111
Connection to 192.168.1.144 2111 port [tcp/kx] succeeded

Below is the log file

terry@terry-All-Series:~$ roslaunch sick_tim sick_tim551_2050001.launch
... logging to /home/terry/.ros/log/71253456-9e59-11e5-a9d4-000d61129c5d/roslaunch-terry-All-Series-28138.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://terry-All-Series:38380/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: jade
 * /rosversion: 1.11.16

NODES
  /
    robot_state_publisher (robot_state_publisher/state_publisher)
    sick_tim551_2050001 (sick_tim/sick_tim551_2050001)

auto-starting new master
process[master]: started with pid [28153]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 71253456-9e59-11e5-a9d4-000d61129c5d
process[rosout-1]: started with pid [28166]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [28169]
process[sick_tim551_2050001-3]: started with pid [28182]
[ERROR] [1449654269.629814527]: No SICK TiM devices connected!
[FATAL] [1449654269.629897819]: Failed to init device: 1
[ERROR] [1449654270.630098409]: LIBUSB - device not open

SOPAS - Error stopping streaming scan data!
sick_tim driver exiting.
[ERROR] [1449654270.673940874]: No SICK TiM devices connected!
[FATAL] [1449654270.673995386]: Failed to init device: 1
^C[sick_tim551_2050001-3] killing on exit
[robot_state_publisher-2] killing on exit

SOPAS - Error stopping streaming scan data!
sick_tim driver exiting.
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

And the launch code is below :

<?xml version="1.0"?>
 <launch> <param name="robot_description" command="$(find xacro)/xacro.py '$(find sick_tim)/urdf/example.urdf.xacro'" /> 
<node name="robot_state_publisher" pkg="robot_state_publisher" type="state_publisher" /> 
<node name="sick_tim551_2050001" pkg="sick_tim" type="sick_tim551_2050001" respawn="false" output="screen"> 
<param name="min_ang" type="double" value="-2.35619449019" />
<param name="max_ang" type="double" value="2.35619449019" /> 
<param name="intensity" type="bool" value="True" /> 
<param name="skip" type="int" value="0" /> 
<param name="frame_id" type="str" value="laser" /> 
<param name="time_offset" type="double" value="-0.001" /> 
<param name="publish_datagram" type="bool" value="False" /> 
<param name="subscribe_datagram" type="bool" value="false" /> 
<param name="time_increment" type="double" value="0.000061722" /> 
<param name="range_min" type="double" value="0.05" /> 
<param name="range_max" type="double" value="10.0" /> 
<param name="hostname" type="string" value="192.168.1.144" /> 
<param name="port" type="string" value="2111" /> 
<param name="timelimit" type="int" value="5" /> 
  
 </node> </launch>

jspricke, 2015-12-09:
Hi Terry,

* TerryGP <notifications@github.com> [2015-12-09 02:09]:
> terry@terry-All-Series:~$ roslaunch sick_tim sick_tim551_2050001.launch

Are you sure this is file? It's usually better to make a local copy of
it and modify that one.

> [ERROR] [1449654269.629814527]: No SICK TiM devices connected!
> [FATAL] [1449654269.629897819]: Failed to init device: 1
> [ERROR] [1449654270.630098409]: LIBUSB - device not open

This is all coming from the USB version.

> <param name="hostname" type="string" value="192.168.1.144" /> 

This seems to be right, can you check that it's actually set using:

```
rosparam get /sick_tim551_2050001/hostname
```

Cheers Jochen

TerryGP, 2015-12-09:
Hi Jochen,

I followed the tutorial of sick package in the ROS website and install it.

I also follow the method to modifiy the IP address and port number and run it, but it just show 

No SICK TiM devices connected!
Failed to init device:1 

I really want to solve this problem and I think I need to state all my steps of installation of ubuntu and Jade, maybe I lost some steps that someone knows. I would be very grateful if someone could give me direction.

I installed the ubuntu 14.04 LTS and after that, I installed the Jade of ROS which I followed website as here :  http://wiki.ros.org/jade/Installation/Ubuntu and ran the "sudo apt-get install ros-jade-sick-tim" to make sure that the installation is completely installed.

When all the installation is set up, i go to the launch folder to find out the sick_tim_551_2050001.launch file to modify the IP address and port number.

After all, I use the command "roslaunch sick_tim sick_tim551_2050001.launch" to run the program, but it not work until now. Did I miss some steps?

jspricke, 2015-12-09:
Please do:
```
roslaunch path/to/file/you/edited/sick_tim551_2050001.launch
```

TerryGP, 2015-12-09:
http://wiki.ros.org/sick_tim

Thanks & regards.

學生 蘇登峰
國立台北科技大學－車輛工程系
Student Terry Su
Department of Vehicle Engineering,Taipei Tech
Phone: +886-2-27712171 ext.3672
Mobile: +886-9-58023555

TerryGP, 2015-12-09:
Hi Jochen,
Thank you so much from my heart, it connects to the lidar!  
[ INFO] [1449666439.051912931]: Waiting 50 seconds for device to connect.
[ INFO] [1449666439.052682443]: Succesfully connected to 192.168.1.144:2111

I have the other problem which is about the lidar detection display as same as the bottom of this website:
http://wiki.ros.org/sick_tim
Can you show the real time detection display as the website?
And did you get lidar's detection files? 
Thank you so much for your kindly help about the connection sincerely.

jspricke, 2015-12-09:
The screen shot is done using [rviz](http://wiki.ros.org/rviz).

TerryGP, 2015-12-09:
I got it! I should install it to show the real time detection. 
And do you get the all data of distance from the lidar?

jspricke, 2015-12-09:
This is a bug tracker, not a message board. Please don't ask support questions.
Sure we get all distance values from the lidar.

TerryGP, 2015-12-09:
OK, thank you for your kindly help and reminding.

TerryGP, 2015-12-10:
Hello,
I have a problem is about shows the detection by rviz. I am using ROS Jade version of the Sick Tim5XX stack with a sick_tim551_2050001(Sick part number : 1060445). And my OS is ubuntu 14.04LTS and running with ROS Jade. I can connect to my sick lidar successfully, but I can't get any display in the rviz now.

I followed the steps of installation for the rviz as following link:
http://wiki.ros.org/rviz/UserGuide

Because my lidar just support the Jade of ROS, I change the command of installation. As below are the commands that I typed in the terminal: 
sudo apt-get install ros-jade-rviz 
rosdep install rviz source 
source /opt/ros/jade/setup.bash 
roscore & rosrun 
rviz rviz

And there is a window of rviz pop out, but without the detection of my lidar. As below is the link of picture of the rviz:
https://drive.google.com/file/d/0B6iIqQqNZmCQbjJqX3J3RzRaN2JmTDZrNzlFRnpGMG5KeXpJ/view

It doesn't show my lidar's detection(my lidar is a 2D laser scanner).
I would be grateful if someone can give me some direction.

jspricke, 2015-12-10:
Please stop using this issue tracker to ask support questions!
If you want to learn ROS, start here: http://wiki.ros.org/ROS/Tutorials

mintar, 2015-12-10:
I agree with @jspricke , especially because your new question is not about sick_tim, but about rviz. Also feel free to ask questions on http://answers.ros.org/questions/.