uos/sick_tim #29: TIM571-2050101 LIBUSB - Read Error: -7 or -9

ahcorde, 2015-10-14:
Hi, 

I'm trying to work with the SICK laser TIM571-2050101. But I always received two different errors:
 
**LIBUSB - Read Error: -7 (LIBUSB_ERROR_TIMEOUT)** or **LIBUSB - Read Error: -9 ( LIBUSB_ERROR_PIPE )**

The error is in the function libusb_bulk_transfer of libusb

```
[ERROR] [1444834004.223997555]: LIBUSB - Read Error: -7.
[ERROR] [1444834004.224061386]: SOPAS - Error reading variable 'DeviceIdent'.
[ERROR] [1444834005.225027697]: LIBUSB - Read Error: -7.
[ERROR] [1444834005.225102733]: SOPAS - Error reading variable 'SerialNumber'.
[ERROR] [1444834006.426673049]: LIBUSB - Read Error: -7.
[ERROR] [1444834006.426734884]: SOPAS - Error reading variable 'FirmwareVersion'.
[ERROR] [1444834007.628947346]: LIBUSB - Read Error: -7.
[ERROR] [1444834007.629010369]: SOPAS - Error starting to stream 'LMDscandata'.
[FATAL] [1444834007.629112005]: Failed to init scanner: 1
```

@vmayoral

vmayoral, 2015-10-14:
+1 on this, reproduced it in another machine and the driver also seems to be faulty. My log:
```
victor@ubuntu:/opt/ros/indigo$ roslaunch sick_tim sick_tim551_2050001.launch ... logging to /home/victor/.ros/log/51accd88-7284-11e5-be21-001c42d5b830/roslaunch-ubuntu-4393.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:55029/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.13

NODES
  /
    robot_state_publisher (robot_state_publisher/state_publisher)
    sick_tim551_2050001 (sick_tim/sick_tim551_2050001)

auto-starting new master
process[master]: started with pid [4408]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 51accd88-7284-11e5-be21-001c42d5b830
process[rosout-1]: started with pid [4421]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [4438]
process[sick_tim551_2050001-3]: started with pid [4439]
[ INFO] [1444834834.029360398]: SICK AG - TIM3XX - [1]
[ INFO] [1444834834.029451400]: ----------------------------------------
[ INFO] [1444834834.029530042]: Device Class: 0x0
[ INFO] [1444834834.029578884]: VendorID:     0x19a2
[ INFO] [1444834834.029621304]: ProductID:    0x5001
[ INFO] [1444834834.029662468]: ----------------------------------------
[ INFO] [1444834834.029709216]: Interfaces: 1
[ INFO] [1444834834.029737778]: ----------------------------------------
[ INFO] [1444834834.029779152]: Number of alternate settings: 1
[ INFO] [1444834834.029819389]: Interface number: 0
[ INFO] [1444834834.029858828]: Number of endpoints: 2
[ INFO] [1444834834.029897988]: Descriptor Type: 5
[ INFO] [1444834834.029948392]: EP Address: 129
[ INFO] [1444834834.029962602]: Descriptor Type: 5
[ INFO] [1444834834.029973949]: EP Address: 2
[ INFO] [1444834834.029986367]: ----------------------------------------
libusbx: error [_get_usbfs_fd] libusbx couldn't open USB device /dev/bus/usb/001/004: Permission denied
libusbx: error [_get_usbfs_fd] libusbx requires write access to USB device nodes.
[ERROR] [1444834834.030054786]: LIBUSB - Cannot open device; please read sick_tim/udev/README
[FATAL] [1444834834.030104432]: Failed to init device: 1
[ERROR] [1444834835.030856842]: LIBUSB - device not open

SOPAS - Error stopping streaming scan data!
sick_tim driver exiting.
[ INFO] [1444834835.056475633]: SICK AG - TIM3XX - [1]
[ INFO] [1444834835.056511833]: ----------------------------------------
[ INFO] [1444834835.056521409]: Device Class: 0x0
[ INFO] [1444834835.056528495]: VendorID:     0x19a2
[ INFO] [1444834835.056534893]: ProductID:    0x5001
[ INFO] [1444834835.056541359]: ----------------------------------------
[ INFO] [1444834835.056550965]: Interfaces: 1
[ INFO] [1444834835.056558025]: ----------------------------------------
[ INFO] [1444834835.056564774]: Number of alternate settings: 1
[ INFO] [1444834835.056571159]: Interface number: 0
[ INFO] [1444834835.056578155]: Number of endpoints: 2
[ INFO] [1444834835.056779267]: Descriptor Type: 5
[ INFO] [1444834835.056826339]: EP Address: 129
[ INFO] [1444834835.056869013]: Descriptor Type: 5
[ INFO] [1444834835.056910491]: EP Address: 2
[... 55 lines cut ...]
SOPAS - Error stopping streaming scan data!
sick_tim driver exiting.
[ INFO] [1444834838.132644889]: SICK AG - TIM3XX - [1]
[ INFO] [1444834838.132733899]: ----------------------------------------
[ INFO] [1444834838.132775965]: Device Class: 0x0
[ INFO] [1444834838.132812818]: VendorID:     0x19a2
[ INFO] [1444834838.132848747]: ProductID:    0x5001
[ INFO] [1444834838.132882783]: ----------------------------------------
[ INFO] [1444834838.132919464]: Interfaces: 1
[ INFO] [1444834838.132954093]: ----------------------------------------
[ INFO] [1444834838.132988349]: Number of alternate settings: 1
[ INFO] [1444834838.133022161]: Interface number: 0
[ INFO] [1444834838.133055915]: Number of endpoints: 2
[ INFO] [1444834838.133088994]: Descriptor Type: 5
[ INFO] [1444834838.133121767]: EP Address: 129
[ INFO] [1444834838.133155246]: Descriptor Type: 5
[ INFO] [1444834838.133188504]: EP Address: 2
[ INFO] [1444834838.133222202]: ----------------------------------------
libusbx: error [_get_usbfs_fd] libusbx couldn't open USB device /dev/bus/usb/001/004: Permission denied
libusbx: error [_get_usbfs_fd] libusbx requires write access to USB device nodes.
[ERROR] [1444834838.133271510]: LIBUSB - Cannot open device; please read sick_tim/udev/README
[FATAL] [1444834838.133313182]: Failed to init device: 1
^C[sick_tim551_2050001-3] killing on exit

```

jspricke, 2015-10-14:
Do you have access to the device file (`/dev/bus/usb/001/004`)? Make sure you followed the instructions given here: https://github.com/uos/sick_tim/tree/indigo/udev.

vmayoral, 2015-10-14:
@jspricke redid it as root user, i've updated the log. That's not the problem.

jspricke, 2015-10-14:
Ah, you confused me with updating the complete comment, I usually prefer to do a new one with more information and use the edit button only to fix spelling.
Also I see that you are using the wrong launch file, there is one for the `sick_tim571_2050101.launch` already, but I don't think that's the problem we see. Sick gave as a scanner as well at ROSCon, but we haven't tried it. Did you try it using TCP/IP?

@mintar do you have time tomorrow to test it or give it to me?

ahcorde, 2015-10-14:
No, we didn't try using TCP/IP only with the USB.

@vmayoral launched with the wrong launch file. But I have compiled the github repository and I have tried with the correct launch file and the result is the same.

jspricke, 2015-10-14:
Ah, there was a problem with the device name containing spaces (or no spaces) at some point. Did you change it?

vmayoral, 2015-10-14:
@jspricke how can we change that?

jspricke, 2015-10-14:
I think with the Windows software coming with it

mintar, 2015-10-15:
It's not the device name, since the problem happens before it even started to stream device data. The device name can only be a problem during parsing of the laser data.

Let's look into this later today with our scanner.

mintar, 2015-10-15:
We just tested it with our scanner, and it works with the `sick_tim571_2050101.launch` file out of the box, both with USB and TCP. Since you tried it on different machines, it seems to be a hardware problem with your scanner. The log shows that your machine could connect to the scanner, but then something went wrong. Can you re-check that you got the power supply, USB cable etc. right?

Sorry for not being more helpful.

jspricke, 2015-10-15:
An other option would be to try TCP/IP.

SICK-NorthAmerica, 2015-10-15:
We haven't seen this issue appear with other TiM571 units but, if it's determined to be a hardware problem, let us know and we'll swap the scanners and try to resolve the issue in-house.

@SICK-Global

mintar, 2015-10-16:
Wow, that's a generous offer by SICK! :)

Before you take them up on it, I'd suggest you try to really make sure it's not a flaky power supply or broken USB cable. One other thing you could try is whether you can get distance measurements from the SICK SOPAS Windows software; that could rule out that it's really not our driver that's at fault.

ahcorde, 2015-10-16:
First at all, thank you for your support. 

Today I have checked the SICK SOPAS Windows software. I'm not able to connect with device. As you can see in the following picture. Windows recognizes the device (Device Manager) but the program can't connect with the device because it isn't able to find it.

[image]

I haven't tried to connect the Laser with the Ethernet, because we need to connect the laser via USB.

It's the first time that I use the SICK SOPAS software if anyone have any kind of suggestions it's more than welcome.

vmayoral, 2015-10-16:
@SICK-NorthAmerica, @SICK-Global, @mintar thanks for the kind support. 

We're aiming here to provide official support for SICK sensors in our artificial brains (http://erlerobotics.com/docs/Artificial_Brains/Erle-Brain2/index.html) and robots. We've tried different cables and Operating Systems and so far no luck. 
>An other option would be to try TCP/IP.

I'm willing to give this a try. We are using USB for now,  Where can we get the appropriate Ethernet cable to test this out?

mintar, 2015-10-16:
The appropriate Ethernet cable should be [this one](https://mysick.com/partnerPortal/ProductCatalog/DataSheet.aspx?ProductID=81290#). You can get it either from SICK or probably from the vendor where you got your power adapter cable, I guess.

ahcorde, 2015-10-30:
We have just received and test the Ethernet cable and the laser is working fine.

We have documented how it works in Erle-Brain 2 with ROS [1]. 

If someone have any ideas why the USB interfaces it isn't working it's most than welcome ;) 

[1] http://erlerobotics.com/Docs/Artificial_Brains/Additional_Devices/SICK_Laser_Scan

Thank you everyone for your help

mintar, 2015-10-30:
Great to hear that it's now working for you!

closed 2015-10-30 by mintar

garyservin, 2015-10-31:
I'm having the same issue with my TiM571 using USB, unfortunately I don't have access to an Ethernet cable so I can't test it. It was working fine until today.

### dmesg ouput
[ 1305.128734] usb 2-3: new high-speed USB device number 18 using xhci_hcd
[ 1305.264060] usb 2-3: New USB device found, idVendor=19a2, idProduct=5001
[ 1305.264072] usb 2-3: New USB device strings: Mfr=0, Product=2, SerialNumber=3
[ 1305.264073] usb 2-3: Product: TIM3xx
[ 1305.264074] usb 2-3: SerialNumber: 15340291

### lsusb -v
Bus 002 Device 018: ID 19a2:5001  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x19a2 
  idProduct          0x5001 
  bcdDevice            1.09
  iManufacturer           0 
  iProduct                2 TIM3xx
  iSerial                 3 15340291
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          1 Default Configuration
    bmAttributes         0xc0
      Self Powered
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              4 SICK SOPAS
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0001
  Self Powered

garyservin, 2015-10-31:
**Update** got it to work on another ubuntu machine I have, which hasn't been updated in a while (at least one month).

So this may be an issue with some package that has been updated. I've checked libsusb but both machines have the same version. Which other packages should I take a look at?

jspricke, 2015-11-01:
@garyservin: Which version(s) of Ubuntu did you try? Did you check the access rights of the device file?

garyservin, 2015-11-01:
Both machines have Lubuntu 14.04.

I'm using the latest driver pulled from github and built yesterday with the same udev rules installed.

I have permissions for the plugdev group for both users in both machines. I have checked the permissions for the device file and they are the same on both machines.

I've also tried running the driver as root and it still didn't work.

garyservin, 2015-11-01:
**New update: It works!**

I've tried some more stuff, including downgrading the kernel, but I couldn't get the driver to work.

What I did was use the ROS' shadow repo, which I was already using on the other machine. I'm not sure which exact package fixed the issue though.

libazhahei, 2023-08-28:
I have the same situation with you, I've found the requests seems to diff from the terminal on SOPAS. 
The command of read DeviceIdent should be "sRI 0" rather than  the original version "sRI0" which is at the sick_tim_common.cpp