uos/sick_tim #30: No frame update on sick_tim551

thomassimon, 2015-11-06:
Hi,
I think there is a missing param to change the defaut name of the laser frame.
I have change the param in the launch file but there is no change.

mintar, 2015-11-10:
Here is how you do it:

    <launch>
      <node name="sick_tim551_2050001" pkg="sick_tim" type="sick_tim551_2050001" respawn="false" output="screen">
          <param name="frame_id" type="str" value="my_laser" />
      </node>
    </launch>

mintar, 2015-11-10:
Oh, and you can check that it's working by running `rostopic echo /scan` and checking the `frame_id`.

closed 2015-11-10 by mintar

thomassimon, 2015-11-10:
Indeed on the topic, it's the correct frame.
But I have checked on the tf monitor and the tf tree.
Also I use the launch file of the sick_tim551.

jspricke, 2015-11-10:
If it's wrong in the tf tree, it's on your side, check your URDF.
The modifications to the launch file proposed by mintar are for the sick_tim551.

thomassimon, 2015-11-10:
For the example, do I have to change it in the robot_description and the urdf linked to the package ?

jspricke, 2015-11-10:
If you use our urdf macro, change it like: `<xacro:sick_tim310 name="my_laser" ros_topic="scan" />`, name is the tf name.