uos/sick_tim #31: Sick TIM310 - LIBUSB - Read Error: -7

JoaquimUdeM, 2016-01-21:
Hi, I am having issues getting my Sick Tim310-1030000 to publish data on /scan.

I initially I had a TIMEOUT error, I fixed that by increasing the time out by 500ms in sick_tim_common_usb.cpp. 

LIBUSB_ERROR_TIMEOUT + 500

Now I have another error:

```
... logging to /home/ponn/.ros/log/b15e236c-c068-11e5-bd53-dc5360922b79/roslaunch-ponn-22050.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://139.103.89.70:58715/

SUMMARY
========

PARAMETERS
 * /robot_description
 * /rosdistro
 * /rosversion

NODES
  /
    robot_state_publisher (robot_state_publisher/state_publisher)
    sick_tim310 (sick_tim/sick_tim310)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[robot_state_publisher-1]: started with pid [22071]
process[sick_tim310-2]: started with pid [22089]
[ INFO] [1453404496.214508875]: SICK AG - TIM3XX - [1]
[ INFO] [1453404496.214598429]: ----------------------------------------
[ INFO] [1453404496.214641661]: Device Class: 0x0
[ INFO] [1453404496.214676574]: VendorID:     0x19a2
[ INFO] [1453404496.214714734]: ProductID:    0x5001
[ INFO] [1453404496.214748722]: ----------------------------------------
[ INFO] [1453404496.214818428]: Interfaces: 1
[ INFO] [1453404496.214855117]: ----------------------------------------
[ INFO] [1453404496.214897265]: Number of alternate settings: 1
[ INFO] [1453404496.214933093]: Interface number: 0
[ INFO] [1453404496.214965705]: Number of endpoints: 2
[ INFO] [1453404496.214998357]: Descriptor Type: 5
[ INFO] [1453404496.215033398]: EP Address: 129
[ INFO] [1453404496.215064303]: Descriptor Type: 5
[ INFO] [1453404496.215095864]: EP Address: 2
[ INFO] [1453404496.215131140]: ----------------------------------------
[ INFO] [1453404496.215206142]: LIBUSB - Claimed interface
[ERROR] [1453404497.220469100]: LIBUSB - Read Error: -7.
[ERROR] [1453404497.220819113]: Read Error when getting datagram: -7.

SOPAS - Stopped streaming scan data.
LIBUSB - Released Interface.
```

I was wondering if anyone knows the solution to this error.

The error occured on both of my out of the box SICK Tim-lasers

Thanks,
Joaquim.

mintar, 2016-01-22:
FYI, "Read error -7" is also LIBUSB_ERROR_TIMEOUT.

There are a couple of related (closed) issues, perhaps you find some useful info there: #7 #29 

Unfortunately, I won't have time in the next week for this. @jspricke , could you take a look?

jspricke, 2016-01-22:
Hi Joaquim,

Do you have the scanner connected through a hub, or directly to the PC? Can you try it on a different computer? Can you the steps Martin proposed here: https://github.com/uos/sick_tim/issues/7#issuecomment-32094778 and send me the bag file?

mintar, 2016-03-30:
Closing due to lack of response.

closed 2016-03-30 by mintar

mintar, 2016-03-31:
This bug probably has the same cause as #33 and #35 : The firmware is probably >= 2.50, which doesn't support ranging.