uos/sick_tim #32: Not getting intensities from Sick 571 / TCP

andsmith, 2016-01-26:
The .intensities attributes of the LaserScan messages is empty, even though the parameter is true:

     $ rosparam get /sick_tim571_2050101/intensity 
     true

And my launch file explicitly sets it:

     <?xml version="1.0"?>
     <launch>
       <param name="robot_description" command="$(find xacro)/xacro.py '$(find sick_tim)/urdf/example.urdf.xacro'" />
       <node name="robot_state_publisher" pkg="robot_state_publisher" type="state_publisher" />

       <node name="sick_tim571_2050101" pkg="sick_tim" type="sick_tim551_2050001" respawn="false" output="screen">
           <param name="min_ang" type="double" value="-2.35619449019" />
           <param name="max_ang" type="double" value="2.35619449019" />
           <param name="intensity" type="bool" value="True" />
           <param name="skip" type="int" value="0" />
           <param name="frame_id" type="str" value="laser" />
           <param name="time_offset" type="double" value="-0.001" />
           <param name="publish_datagram" type="bool" value="False" />
           <param name="subscribe_datagram" type="bool" value="false" />
           <param name="range_min" type="double" value="0.05" />
           <param name="range_max" type="double" value="25.0" />

              <param name="hostname" type="string" value="192.168.0.68" />
              <param name="port" type="string" value="2112" />
              <param name="timelimit" type="int" value="5" />
    
       </node>
     </launch>

(the ranges & other data look correct, though)

I did some poking around in the ROS node source, and it appears the RSSI data aren't even in the datagrams it's getting from the device.  

Could this be a malfunctioning device, or maybe incorrectly configured?

procopiostein, 2016-01-26:
Same problem here. Therefore, I do not think it is a malfunctioning device.

SICK-NorthAmerica, 2016-01-26:
To better manage data rates for a wide variety of customers, the TiM does not output RSSI data by default.  It must first be enabled in the configuration software by downloading [SOPAS-ET 3.2.0](https://www.mysick.com/eCat.aspx?go=FinderSearch&Cat=Row&At=Fa&Cult=English&FamilyID=235&Category=Software&Selections=54195,54293), connecting to the sensor using USB or Ethernet, logging in (UID: Authorized Client, PWD: client), and clicking the checkbox shown below:

[image]

SICK-NorthAmerica, 2016-01-26:
Although your line *<param name="intensity" type="bool" value="True" />* should perform the same function.  Running the configuration software may be a good way to double-check that the parameter is properly set.

mintar, 2016-01-26:
The parameter `intensity` only works if the device is already configured to output RSSI. If the datagrams don't include RSSI data, the parameter does nothing. So the solution is to enable RSSI output in SOPAS, as described by @SICK-NorthAmerica . I'll add it to the ROS wiki page!

andsmith, 2016-01-26:
Thanks guys!  It was the configuration indeed.  It's working now.

closed 2016-01-26 by andsmith

mintar, 2016-01-27:
Actually, I've checked the code again. The `intensity` param didn't do anything in the TiM551/571 parser, only for the other scanners. I've changed that and added a (hopefully helpful) warning.

What I said before remains valid: Intensities are only output if `intensity = true` *and* RSSI output is enabled in the scanner.

procopiostein, 2016-01-28:
thanks, it worked!