uos/sick_tim #33: Sick TIM351-2134001 not functional on latest firmware 2.5

metallo25, 2016-02-05:
Hello,
I have been using this package for sick tim 351 for a while. Its been working great with previous firmware 2.3 and 2.4, but recently 2.5 does not give the data gram packets. I have checked that the lidar works well in windows side and its fully functional. 

Here is the error I get: 
```
[ERROR] [1454715021.672259907]: sendSOPASCommand: failed attempt to read from socket: 125: Operation canceled
[ERROR] [1454715021.672337516]: get_datagram: no data available for read after 1000 ms
```
Please let me know about this.

jspricke, 2016-02-06:
Interesting, I haven't updated a scanner to 2.5 up to now. Can you try it over USB? Can you try to sniff what's going on using wireshark? I won't have much time next week, so it could take some time for me to replicate this over here.

SICK-NorthAmerica, 2016-02-08:
This is not an issue with the package but, rather, a limitation of the most recent TiM3xx firmware.  Please contact me at aaron.rothmeyer@sick.com to talk about the application and a possible fix.

mintar, 2016-02-09:
Thanks for taking care of this, @SICK-NorthAmerica ! If there is any information about that we should add to the README, let us know!

SICK-NorthAmerica, 2016-02-12:
The TiM351 is a detection-only device meaning that it should only output GPIO-style digital outputs based on the presence or absence of an object in a preconfigured zone.  A firmware bug present on versions prior to V2.50 allowed ranging data output over TCP and USB when that functionality should be limited to the TiM5xx series.  To prevent future confusion, I'd recommend modifying the wiki to either remove the "TiM351-2134001" from the list or note "firmware revisions prior to V2.50".  If a user of the sick_tim package is using BOTH ranging and detection modes of the TiM351 in a single application, they should contact me at [aaron.rothmeyer@sick.com](mailto:aaron.rothmeyer@sick.com) to let me know.

jspricke, 2016-02-12:
I've added a note to the ROS wiki, thanks for the explanation @SICK-NorthAmerica.

closed 2016-02-12 by jspricke

akrv, 2018-05-30:
Hi,
Laser scanner mode: TiM571

is this problem related to this same issue?

I have attached a pic of what is happening.
the second picture, is the above error, which happens after some time.
If not, is there is an answer why it is happening like this?
(didnt want to start a new issue, since the problem seemed the same)

[image]

[image]

jspricke, 2018-05-30:
Don't know, you should find your version number in the Sick software..

Jyck, 2019-08-18:
its all working very well after some changes on my Tim351 with firmware V2.61

Sopas Command "LMDscandata" is not working on >V2.5 (maybe deactivated by sick to buy a more expensive scanner)

so just look how Sopas connect to the scanner and rebuild it.

here are the important changes:

stop streaming command: 
old command:  "\x02sEN LMDscandata 0\x03\0"
new command: "\x02sEI 5B 0\x03\0"

start streaming:
old command:  "\x02sEN LMDscandata 1\x03\0"
new command:  "\x02sEI 5B 1\x03\0"  direct followed by "\x02sRI E7\x03\0"

all can be found in sick_tim_common.cpp    (line 78, 245) 
make sure to change/skip the isCompatibleDevice (Line 270)

greetings

jspricke, 2019-08-18:
Nice find!

could you send a pull request with the changes?

Jyck, 2019-08-18:
yes sure, but time-limited only in a few days

lee6225, 2022-07-13:
After the product that was operating in tim351 v2.31 was changed to V2.51 by product replacement, sopas checked the operation, but it does not work due to a communication error in the actual connected system.
Can you advise me

SICK-NorthAmerica, 2022-07-19:
> After the product that was operating in tim351 v2.31 was changed to V2.51 by product replacement, sopas checked the operation, but it does not work due to a communication error in the actual connected system. Can you advise me Can you send me v2.31 firmware?

@lee6225 Please reach out to me directly at aaron.rothmeyer@sick.com.  I'm the product manager for the TiM line at SICK and I should be able to assist if you are still seeing this issue.  Thank you!

hoanganhidi0822, 2024-08-08:
> its all working very well after some changes on my Tim351 with firmware V2.61
>
> Sopas Command "LMDscandata" is not working on >V2.5 (maybe deactivated by sick to buy a more expensive scanner)
>
> so [...]

> its all working very well after some changes on my Tim351 with firmware V2.61
>
> Sopas Command "LMDscandata" is not working on >V2.5 (maybe deactivated by sick to buy a more expensive scanner)
>
> so [...]

I have the same issue with sick TiM310 v5.10, can you help me

hoanganhidi0822, 2024-08-08:
> > After the product that was operating in tim351 v2.31 was changed to V2.51 by product replacement, sopas checked the operation, but it does not work due to a communication error in the actual connected system. Can you advise me Can you send me v2.31 firmware?
> 
> @lee6225 Please reach out to me directly at [aaron.rothmeyer@sick.com](mailto:aaron.rothmeyer@sick.com). I'm the product manager for the TiM line at SICK and I should be able to assist if you are still seeing this issue. Thank you!

I sent you an email last night. I can't read data from lidar using "sRN LMDscandata" command. Please help me