uos/sick_tim #34: TIM551 Ethernet connection error Ubuntu 14.04

felixmn91, 2016-03-03:
Everything works fine under windows with the SOPA Engineering Tool.

After changing the IP address in the launch file and un-commenting the Ethernet section, I get under Ubuntu this error:

[image]

I thought that perhaps disabling networking changes anything, but then I get this error: 

[image]

Would you please help me?

Felix

mintar, 2016-03-04:
Plesse attach the launch file you're using.
-- 
Diese Nachricht wurde von meinem Android-Mobiltelefon mit K-9 Mail gesendet.

mintar, 2016-03-04:
Also make sure that you create a new wired network connection in NetworkManager and put in the following settings:

[image]

You can use any address like `169.254.104.*`, but **not** `169.254.104.128` (that's the IP of your scanner).

You can check if that worked by running `ping 169.254.104.128`.

felixmn91, 2016-03-04:
Mintar, thank you so much!! It is working now :)

mintar, 2016-03-04:
Glad to hear! :)

closed 2016-03-04 by mintar