uos/sick_tim #35: Cannot receive scans or datagram from Tim310

doisyg, 2016-03-31:
Hello,
I cannot reveive laser scans nor datagram from a brand new Tim310, maybe i am missing an obvious confuguration step
I ran:` roslaunch sick_tim310.launch ` and get the following console output:

```
[ INFO] [1459412901.374274450]: SICK AG - TIM3XX - [1]
[ INFO] [1459412901.374340031]: ----------------------------------------
[ INFO] [1459412901.374357500]: Device Class: 0x0
[ INFO] [1459412901.374375867]: VendorID:     0x19a2
[ INFO] [1459412901.374397504]: ProductID:    0x5001
[ INFO] [1459412901.374418771]: ----------------------------------------
[ INFO] [1459412901.374444977]: Interfaces: 1
[ INFO] [1459412901.374464671]: ----------------------------------------
[ INFO] [1459412901.374484516]: Number of alternate settings: 1
[ INFO] [1459412901.374503842]: Interface number: 0
[ INFO] [1459412901.374522273]: Number of endpoints: 2
[ INFO] [1459412901.374540360]: Descriptor Type: 5
[ INFO] [1459412901.374558490]: EP Address: 129
[ INFO] [1459412901.374575574]: Descriptor Type: 5
[ INFO] [1459412901.374592462]: EP Address: 2
[ INFO] [1459412901.374612250]: ----------------------------------------
[ INFO] [1459412901.374668622]: LIBUSB - Claimed interface
[ WARN] [1459412902.380436408]: LIBUSB - Read Error: LIBUSB_ERROR_TIMEOUT.
[ WARN] [1459412903.381279210]: LIBUSB - Read Error: LIBUSB_ERROR_TIMEOUT.
[ WARN] [1459412904.382005495]: LIBUSB - Read Error: LIBUSB_ERROR_TIMEOUT.
[ WARN] [1459412905.382584076]: LIBUSB - Read Error: LIBUSB_ERROR_TIMEOUT.
[ WARN] [1459412906.383259932]: LIBUSB - Read Error: LIBUSB_ERROR_TIMEOUT.
[ WARN] [1459412907.384010439]: LIBUSB - Read Error: LIBUSB_ERROR_TIMEOUT.
[ WARN] [1459412908.384579540]: LIBUSB - Read Error: LIBUSB_ERROR_TIMEOUT.
[ WARN] [1459412909.385220461]: LIBUSB - Read Error: LIBUSB_ERROR_TIMEOUT.
[ WARN] [1459412910.385830455]: LIBUSB - Read Error: LIBUSB_ERROR_TIMEOUT.
[ WARN] [1459412911.386313384]: LIBUSB - Read Error: LIBUSB_ERROR_TIMEOUT.
```

Nothing is published on the topics /scan and /datagram (I have set publish_datagram to true).
The usb connection seems okay though, as i can somehow see the Serial Number of my device in the diagnotic topic, here is some echo of /diagnostics:

```
header: 
  seq: 17
  stamp: 
    secs: 1459415111
    nsecs: 247821658
  frame_id: ''
status: 
  - 
    level: 2
    name: sick_tim310: /scan topic status
    message: LIBUSB - Read Error: LIBUSB_ERROR_TIMEOUT.
    hardware_id: ''
    values: []
---
header: 
  seq: 18
  stamp: 
    secs: 1459415111
    nsecs: 247946211
  frame_id: ''
status: 
  - 
    level: 2
    name: sick_tim310: /scan topic status
    message: No events recorded.; No data since last update.
    hardware_id: sRA 0 6 TiM31x E V2.51-21.08.15 sRA SerialNumber 8 16100082
    values: 
      - 
        key: Events in window
        value: 0
      - 
        key: Events since startup
        value: 0
      - 
        key: Duration of window (s)
        value: 9.011856
      - 
        key: Actual frequency (Hz)
        value: 0.000000
      - 
        key: Target frequency (Hz)
        value: 15.000000
      - 
        key: Minimum acceptable frequency (Hz)
        value: 13.500000
      - 
        key: Maximum acceptable frequency (Hz)
        value: 16.500000
      - 
        key: Earliest timestamp delay:
        value: 0.000000
      - 
        key: Latest timestamp delay:
        value: 0.000000
      - 
        key: Earliest acceptable timestamp delay:
        value: -1.000000
      - 
        key: Latest acceptable timestamp delay:
        value: 0.086667
      - 
        key: Late diagnostic update count:
        value: 0
      - 
        key: Early diagnostic update count:
        value: 0
      - 
        key: Zero seen diagnostic update count:
        value: 0
---
header: 
  seq: 19
  stamp: 
    secs: 1459415112
    nsecs: 248420411
  frame_id: ''
status: 
  - 
    level: 2
    name: sick_tim310: /scan topic status
    message: LIBUSB - Read Error: LIBUSB_ERROR_TIMEOUT.
    hardware_id: ''
    values: []
---
header: 
  seq: 20
  stamp: 
    secs: 1459415112
    nsecs: 248683595
  frame_id: ''
status: 
  - 
    level: 2
    name: sick_tim310: /scan topic status
    message: No events recorded.; No data since last update.
    hardware_id: sRA 0 6 TiM31x E V2.51-21.08.15 sRA SerialNumber 8 16100082
    values: 
      - 
        key: Events in window
        value: 0
      - 
        key: Events since startup
        value: 0
      - 
        key: Duration of window (s)
        value: 10.012495
      - 
        key: Actual frequency (Hz)
        value: 0.000000
      - 
        key: Target frequency (Hz)
        value: 15.000000
      - 
        key: Minimum acceptable frequency (Hz)
        value: 13.500000
      - 
        key: Maximum acceptable frequency (Hz)
        value: 16.500000
      - 
        key: Earliest timestamp delay:
        value: 0.000000
      - 
        key: Latest timestamp delay:
        value: 0.000000
      - 
        key: Earliest acceptable timestamp delay:
        value: -1.000000
      - 
        key: Latest acceptable timestamp delay:
        value: 0.086667
      - 
        key: Late diagnostic update count:
        value: 0
      - 
        key: Early diagnostic update count:
        value: 0
      - 
        key: Zero seen diagnostic update count:
        value: 0

```

mintar, 2016-03-31:
The reason is probably that the firmware version you have is too new: V2.51-21.08.15. As explained by @SICK-NorthAmerica 's comment in #33 , the SICK TiM3xx devices were never meant to be used for ranging, only for detection. Unfortunately, there is nothing we can do in the sick_tim driver. You could try contacting @SICK-NorthAmerica directly (his email is in #33 ) and see if there is anything he can do.

doisyg, 2016-03-31:
Thank you very much for your prompt answer. There was an issue with the order, we received a 310 instead of 310s01, we are going to proceed to an exchange.

closed 2016-03-31 by doisyg