uos/sick_tim #36: Driver should detect incompatible firmware versions on 3xx (LIBUSB_ERROR_TIMEOUT)

mintar, 2016-03-31:
The driver should print an error when the device is:

* a TiM 3xx, and
* has firmware >= V2.50 (TODO: check this) .

As mentioned in #33 and #35 , the newer firmware versions don't support ranging, which leads to  "Read error -7" or LIBUSB_ERROR_TIMEOUT error messages

SICK-NorthAmerica, 2016-03-31:
Good idea for an enhancement.  If @SICK-NorthAmerica can be of any help w/ revision lists, etc., let us know!

mintar, 2016-04-11:
Thanks for your offer, @SICK-NorthAmerica ! Indeed it would be very helpful if you could provide a revision list, so that we know which versions of the firmware provide ranging data and which ones don't.

SICK-NorthAmerica, 2016-04-11:
The following links show all official revision numbers for all current TiM models.  If you can include these numbers, it should ensure no TiM goes unrecognized in the future.  The only caveat is that revisions of firmware >= V2.50 for the TiM3xx should print an error indicating that this model/firmware combination does not support ranging output.  Let me know if you have any questions or need any further information.

TiM31x/32x:
[image]

TiM55x:
[image]

TiM56x:
[image]

TiM35x:
[image]

TiM36x:
[image]

PR #37 (2016-04-14): Check for firmware versions without range output
Closes #36 .
commit 11ca91c8 (2016-04-14): Check for firmware versions without range output

Closes #36 .
closed 2016-04-14 by mintar