uos/sick_tim #39: sending  configuration to sick_tim571

FarizaKhelifaoui, 2016-08-25:
Hi,
I'am using ROS to communicate with sick_tim571. I launch roslaunch sick_tim sick_tim571_2050101.launch => Succesfuly connected to device).
I want to change the IP address of the sick_tim device with a ros command (launch an sick_tim...config.launch). someone has an idea how to do it, please?

Fariza

mintar, 2016-08-26:
Sorry, but the sick_tim driver doesn't support changing the IP at the moment. Please use the SopasET Windows tool to do that.

FarizaKhelifaoui, 2016-08-29:
Hi Martin,
You are right. I modify the driver to have the possibilty to do it. If it
works, I will push it.
I don't want to use SopasET windows because I want configure all componants
of the robot when I launch a script.

FarizaKhelifaoui, 2016-08-29:
[(http://192.168.100.2/index.php/apps/files/?dir=%2FConfigurtion%20des%20p%C3%A9riph%C3%A9riques%2FSpecifications%20techniques%20du%20sick_tim)]
The sick tutorial that explains the sick telegrams.

mintar, 2016-08-29:
> (http://192.168.100.2/index.php/apps/files/?dir=%2FConfigurtion%20des%20p%C3%A9riph%C3%A9riques%2FSpecifications%20techniques%20du%20sick_tim)

192.168.100.2 is a local IP, so it's not accessible from the internet. :)

mintar, 2016-08-29:
> Hi Martin, You are right. I modify the driver to have the possibilty to do it. If it works, I will push it. I don't want to use SopasET windows because I want configure all componants of the robot when I launch a script.

That will not work, because you need to have the IP of the scanner in order to connect to it, in order to change its IP. :)

What you are proposing would mean this sequence of events:

1. You give an IP to your scanner using SopasET (for example 192.168.100.10).
2. Inside your launch file, you start `sick_tim` with the parameter `hostname = 192.168.100.10` (otherwise it won't be able to connect to the scanner in order to change its IP).
3. You add a new functionality to change the IP to let's say 192.168.100.23, so sick_tim connects to the scanner on 192.168.100.10, changes its IP to 192.168.100.23, disconnects again and reconnects on 192.168.100.23.
4. The IP setting of the scanner is permanent, so you have to kill everything, edit your launch file so it will connect to 192.168.100.23 next time.

Step 3 + 4 do not make sense to me. Why not change the IP to 192.168.100.23 in the first step? Another advantage is that this only works as long as the new IP is in the same subnet as before. If you want to change the IP to let's say 192.168.5.42, you have to disconnect your host PC's network connection and change your host PC's IP address to some IP in 192.168.5.*. You cannot do that via a launch file.

FarizaKhelifaoui, 2016-08-31:
Oups! Sorry for the local IP...
https://www.sick.com/media/dox/7/27/927/Technical_information_Telegram_Listing_Ranging_sensors_LMS1xx_LMS5xx_TiM5xx_NAV310_LD_OEM15xx_LD_LRS36xx_en_IM0045927.PDF

I dont see the problem. The original IP is 192.168.0.1. I have to do an
roslaunch sick_tim sick_tim_57...launch ip:=192.168.0.1. Then I set the new
IP with my method setIP(192.168.100.23).
I have to do that (not pass through SopasET) otherwise I can't realize my
project :(

mintar, 2016-08-31:
As I wrote, there are two problems:

1. After you change the IP, you won't be able to connect to the scanner. Your host PC still has an IP in the 192.168.0.* range, and you need to change it to 192.168.100.* .
2. The setting is persistent (as far as I know). That means that the next time you start your scanner, it will already have the IP 192.168.100.23, so you don't need to change it.

I still think that because of these reasons, it's easiest to just change the IP directly with SopasET. But if you don't believe me, just go ahead and try implementing what you want.

FarizaKhelifaoui, 2016-08-31:
the function setIP(ip) returnes an boolean received. If received == true :
subprocess.call(['sudo ifconfig eth0 192.168.100.99 netmask
255.255.255.0'], shell=True); This command changes my host PC IP

mintar, 2016-08-31:
Changing the network config from within a ROS node sounds like a horrible idea, but that's just my opinion.

Seems like we've cleared up everything that relates to the sick_tim package. Can we close this issue?

FarizaKhelifaoui, 2016-08-31:
Thank you for your for taking the time to respond and for your opinion :) . Yes we can close this issue!

closed 2016-08-31 by FarizaKhelifaoui