uos/sick_tim #40: massage rate issus

deepzf, 2016-08-27:
i used tim361 and "rostopic echo" to linstenintg laser's message .
as the sepc of Tim361 shows it should report the ranges at 15Hz,but "rostopic echo" only updated about 5Hz,and Rviz updated messages at same rate.
i  tryed to use  SopasET to config tim361 but nothing found, the report rate is an read only pram.
the fireware of my tim361 is 2.42
How can i rise up the update rate in ros

deepzf, 2016-08-28:
SopasET shows the Frequency is 15Hz

mintar, 2016-08-28:
That is weird, I've never had that problem before. Any other hints to solve this? Does the driver print any error messages?

jspricke, 2016-08-28:
Can you try `rostopic hz /scan`?

deepzf, 2016-08-28:
root@raspberrypi:/home/pi# rostopic hz /scan
subscribed to [/scan]
average rate: 4.988
	min: 0.198s max: 0.202s std dev: 0.00164s window: 5
average rate: 4.994
	min: 0.197s max: 0.203s std dev: 0.00182s window: 10
average rate: 4.997
	min: 0.197s max: 0.203s std dev: 0.00167s window: 15
average rate: 4.998
	min: 0.197s max: 0.203s std dev: 0.00167s window: 20
average rate: 4.998
	min: 0.197s max: 0.203s std dev: 0.00169s window: 25
average rate: 4.998
	min: 0.197s max: 0.203s std dev: 0.00165s window: 30

deepzf, 2016-08-28:
root@raspberrypi:/home/pi# roslaunch sick_tim sick_tim551_2050001.launch 
... logging to /root/.ros/log/fc6d9a8c-6d46-11e6-b51d-b827eb57f198/roslaunch-raspberrypi-12471.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberrypi:42284/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /sick_tim551_2050001/time_increment: 5.5556e-05

NODES
  /
    robot_state_publisher (robot_state_publisher/state_publisher)
    sick_tim551_2050001 (sick_tim/sick_tim551_2050001)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[robot_state_publisher-1]: started with pid [12492]
process[sick_tim551_2050001-2]: started with pid [12493]
[ INFO] [1472406861.672840922]: SICK AG - TIM3XX - [1]
[ INFO] [1472406861.673162378]: ----------------------------------------
[ INFO] [1472406861.673258419]: Device Class: 0x0
[ INFO] [1472406861.673330918]: VendorID:     0x19a2
[ INFO] [1472406861.673404303]: ProductID:    0x5001
[ INFO] [1472406861.673474146]: ----------------------------------------
[ INFO] [1472406861.673577010]: Interfaces: 1
[ INFO] [1472406861.673650603]: ----------------------------------------
[ INFO] [1472406861.673725706]: Number of alternate settings: 1
[ INFO] [1472406861.673801851]: Interface number: 0
[ INFO] [1472406861.673877267]: Number of endpoints: 2
[ INFO] [1472406861.673946069]: Descriptor Type: 5
[ INFO] [1472406861.674017214]: EP Address: 129
[ INFO] [1472406861.674084193]: Descriptor Type: 5
[ INFO] [1472406861.674146484]: EP Address: 2
[ INFO] [1472406861.674237525]: ----------------------------------------
[ INFO] [1472406861.674510751]: LIBUSB - Claimed interface
[ WARN] [1472406861.753457616]: hex: 5DC, scanning_freq: 1500, scan_time: 0.066667
[ WARN] [1472406861.754372191]: Intensity parameter is enabled, but the scanner is not configured to send RSSI values! Please read the section 'Enabling intensity (RSSI) output' here: http://wiki.ros.org/sick_tim.
[ WARN] [1472406861.953578684]: hex: 5DC, scanning_freq: 1500, scan_time: 0.066667
[ WARN] [1472406862.153598556]: hex: 5DC, scanning_freq: 1500, scan_time: 0.066667
[ WARN] [1472406862.356201790]: hex: 5DC, scanning_freq: 1500, scan_time: 0.066667
[ WARN] [1472406862.553421009]: hex: 5DC, scanning_freq: 1500, scan_time: 0.066667
ps:warn messages of scanning_freq  is only for debug

jspricke, 2016-08-28:
* deepzf <notifications@github.com> [2016-08-28 10:57]:
> root@raspberrypi:/home/pi# roslaunch sick_tim sick_tim551_2050001.launch 

Ah, you are using a raspberry pi. Can you try it on a desktop computer?

deepzf, 2016-08-29:
Ok,i will try it later,by the way,what is the meaning of 'scanning_freq:1500'

mintar, 2016-08-29:
> Ok,i will try it later,by the way,what is the meaning of 'scanning_freq:1500'

That's the scanning frequency in 1/100 Hz, meaning 15 Hz. This is fixed for all SICK TiMs and cannot be changed.

Also I don't think it's a performance problem caused by the raspberry pi. The `rostopic hz` output looks to be suspiciously close to 5 Hz, so the publication rate is probably limited by the scanner.

My guess is that you set the parameter `skip = 2` (which means, drop 2 out of 3 messages, resulting in 5 Hz). The solution is to set `skip = 0` (for example by using rqt_reconfigure).

deepzf, 2016-08-29:
root@raspberrypi:/home/pi# rosparam dump | grep skip
  skip: 0, time_increment: 5.5556e-05, time_offset: -0.001}

deepzf, 2016-08-29:
I made some tests，maybe the true Frequency ok the massage is only 5Hz due to the firmware

mintar, 2016-08-29:
I think the SopasET software has a scan visualization function. Does it look better there?
-- 
Diese Nachricht wurde von meinem Android-Mobiltelefon mit K-9 Mail gesendet.

deepzf, 2016-08-30:
SopasET shows the same Frequency，so i think it is the firmware issue

deepzf, 2016-08-30:
thank for all

closed 2016-08-30 by deepzf