uos/sick_tim #42: sick_tim310s01 does not work on Ubuntu Mate 16.04/Raspberry Pi 3

KjeldJensen, 2016-10-20:
In a classroom setup we have tried to get the sick_tim310s01 node up an running on a Raspberry Pi 3 running Ubuntu Mate 16.04. The setup works well on a laptop running Ubuntu 16.04 but on the raspi it is apparently unable to communicate with the TIM310 and causes a stack smashing error. (identical files under ros workspace, just moving the usb cable) 

The sick_tim551_2050001 node works well under both configurations, so the detected problem seems to be isolated to the sick_tim310s01 node.

The problem was reproduced independently by two student teams and the lecturer.

Below is the rosout error when running on the raspi 3:

[ INFO] [1477006478.882499078]: SICK AG - TIM3XX - [1]
[ INFO] [1477006478.882742202]: ----------------------------------------
[ INFO] [1477006478.882878503]: Device Class: 0x0
[ INFO] [1477006478.882967096]: VendorID:     0x19a2
[ INFO] [1477006478.883064908]: ProductID:    0x5001
[ INFO] [1477006478.883150689]: ----------------------------------------
[ INFO] [1477006478.883266365]: Interfaces: 1
[ INFO] [1477006478.883350739]: ----------------------------------------
[ INFO] [1477006478.883440114]: Number of alternate settings: 1
[ INFO] [1477006478.883524540]: Interface number: 0
[ INFO] [1477006478.883607873]: Number of endpoints: 2
[ INFO] [1477006478.883692560]: Descriptor Type: 5
[ INFO] [1477006478.883777507]: EP Address: 129
[ INFO] [1477006478.883857507]: Descriptor Type: 5
[ INFO] [1477006478.883935111]: EP Address: 2
[ INFO] [1477006478.884030058]: ----------------------------------------
[ INFO] [1477006478.884250994]: LIBUSB - Claimed interface
[ERROR] [1477006479.885061559]: LIBUSB - Read Error: -7.
[ERROR] [1477006479.886348113]: SOPAS - Error reading variable 'SerialNumber'.
[ERROR] [1477006480.888117057]: LIBUSB - Read Error: -7.
[ERROR] [1477006480.889940222]: SOPAS - Error reading variable 'FirmwareVersion'.
*** stack smashing detected ***: /opt/ros/kinetic/lib/sick_tim/sick_tim310s01 terminated

jspricke, 2016-10-25:
Hi Kjeld,

* Kjeld Jensen <notifications@github.com> [2016-10-20 16:41]:
> [ INFO] [1477006478.884250994]: LIBUSB - Claimed interface
> [ERROR] [1477006479.885061559]: LIBUSB - Read Error: -7.
> [ERROR] [1477006479.886348113]: SOPAS - Error reading variable 'SerialNumber'. [...]

Can you make sure that you have read and write access to the device?
Whats the output of `ls -l /dev/bus/usb/*`

KjeldJensen, 2016-10-26:
Hi Jochen, 

Thanks for looking into this.

I tried both by adding full rw permissions to the device (o+rw) and running as root, same result. Anyway it should work fine without as I have added the user to the plugdev group, and it evidently works fine with the tim551.

dmesg output right after connection of the TIM310

[  337.617417] usb 1-1.3: new high-speed USB device number 5 using dwc_otg
[  337.724821] usb 1-1.3: New USB device found, idVendor=19a2, idProduct=5001
[  337.724839] usb 1-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=3
[  337.724852] usb 1-1.3: Product: TIM3xx
[  337.724865] usb 1-1.3: SerialNumber: 13390097

Output from ls -l /dev/bus/usb/* without TIM310 connected:

crw-rw-r-- 1 root root 189, 0 Oct 21 01:53 001
crw-rw-r-- 1 root root 189, 1 Oct 21 01:53 002
crw-rw-r-- 1 root root 189, 2 Oct 21 01:53 003

Output from ls -l /dev/bus/usb/* with TIM310 connected:

crw-rw-r-- 1 root root    189, 0 Oct 21 01:53 001
crw-rw-r-- 1 root root    189, 1 Oct 21 01:53 002
crw-rw-r-- 1 root root    189, 2 Oct 21 01:53 003
crw-rw-r-- 1 root plugdev 189, 4 Oct 26 09:57 005

/Kjeld

jspricke, 2016-10-28:
@KjeldJensen could you test the pull-request? Because I played around with the [`USB_TIMEOUT`](https://github.com/uos/sick_tim/blob/2feca0dfeccc7db78d7a88173198bde4df4aef6e/include/sick_tim/sick_tim_common_usb.h#L74) as well (but it worked without it for me in the end).

commit dc75c391 (2016-10-29): Update strtok logic. Fixes #42 (#43)
closed 2016-10-29 by mintar