uos/sick_tim #45: Using two SICK lidars creates problems

alexge233, 2017-02-20:
Hello,

First of all thanks for the packages, very useful!
We have two lidars:
1. a **SICK TIM-510** in the rear, 
2. and a **SICK TIM-561** in the front.

However, we've run into a peculiar issue:

**When both lidars are plugged in (USB) then the front lidar (TIM 561) produces warnings** and there is no data read from either device:

```bash
roslaunch sick_tim sick_tim551_2050001.launch

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.6

NODES
  /
    robot_state_publisher (robot_state_publisher/state_publisher)
    sick_tim551_2050001 (sick_tim/sick_tim551_2050001)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[robot_state_publisher-1]: started with pid [23209]
process[sick_tim551_2050001-2]: started with pid [23210]

[ INFO] [1487597677.973621625]: SICK AG - TIM3XX - [1]
[ INFO] [1487597677.973621625]: ----------------------------------------
[ INFO] [1487597677.973621625]: Device Class: 0x0
[ INFO] [1487597677.973621625]: VendorID:     0x19a2
[ INFO] [1487597677.973621625]: ProductID:    0x5001
[ INFO] [1487597677.973621625]: ----------------------------------------
[ INFO] [1487597677.973621625]: Interfaces: 1
[ INFO] [1487597677.973621625]: ----------------------------------------
[ INFO] [1487597677.973621625]: Number of alternate settings: 1
[ INFO] [1487597677.973621625]: Interface number: 0
[ INFO] [1487597677.973621625]: Number of endpoints: 2
[ INFO] [1487597677.973621625]: Descriptor Type: 5
[ INFO] [1487597677.973621625]: EP Address: 129
[ INFO] [1487597677.973621625]: Descriptor Type: 5
[ INFO] [1487597677.973621625]: EP Address: 2
[ INFO] [1487597677.973621625]: ----------------------------------------
[ INFO] [1487597677.973621625]: SICK AG - TIM3XX - [2]
[ INFO] [1487597677.973621625]: ----------------------------------------
[ INFO] [1487597677.973621625]: Device Class: 0x0
[ INFO] [1487597677.973621625]: VendorID:     0x19a2
[ INFO] [1487597677.973621625]: ProductID:    0x5001
[ INFO] [1487597677.973621625]: ----------------------------------------
[ INFO] [1487597677.973621625]: Interfaces: 1
[ INFO] [1487597677.973621625]: ----------------------------------------
[ INFO] [1487597677.973621625]: Number of alternate settings: 1
[ INFO] [1487597677.973621625]: Interface number: 0
[ INFO] [1487597677.973621625]: Number of endpoints: 2
[ INFO] [1487597677.973621625]: Descriptor Type: 5
[ INFO] [1487597677.973621625]: EP Address: 129
[ INFO] [1487597677.973621625]: Descriptor Type: 5
[ INFO] [1487597677.973621625]: EP Address: 2
[ INFO] [1487597677.973621625]: ----------------------------------------
[ INFO] [1487597677.973621625]: LIBUSB - Claimed interface
[ WARN] [1487597678.017621618]: Field 15 of received data is not equal to 0 (D1). Unexpected data, ignoring scan
[ WARN] [1487597678.085621607]: Field 15 of received data is not equal to 0 (C9). Unexpected data, ignoring scan
[ WARN] [1487597678.153621596]: Field 15 of received data is not equal to 0 (B4). Unexpected data, ignoring scan
[ WARN] [1487597678.217621585]: Field 15 of received data is not equal to 0 (FA). Unexpected data, ignoring scan
[ WARN] [1487597678.285621574]: Field 15 of received data is not equal to 0 (7F). Unexpected data, ignoring scan
[ WARN] [1487597678.349621564]: Field 15 of received data is not equal to 0 (83). Unexpected data, ignoring scan
[ WARN] [1487597678.417621553]: Field 15 of received data is not equal to 0 (CB). Unexpected data, ignoring scan
[ WARN] [1487597678.485621542]: Field 15 of received data is not equal to 0 (DC). Unexpected data, ignoring scan
[ WARN] [1487597678.553621531]: Field 15 of received data is not equal to 0 (4B). Unexpected data, ignoring scan
[ WARN] [1487597678.617621521]: Field 15 of received data is not equal to 0 (93). Unexpected data, ignoring scan
[ WARN] [1487597678.685621510]: Field 15 of received data is not equal to 0 (1C). Unexpected data, ignoring scan
[ WARN] [1487597678.749621499]: Field 15 of received data is not equal to 0 (B). Unexpected data, ignoring scan
[ WARN] [1487597678.817621488]: Field 15 of received data is not equal to 0 (DE). Unexpected data, ignoring scan
[ WARN] [1487597678.885621477]: Field 15 of received data is not equal to 0 (B7). Unexpected data, ignoring scan
[ WARN] [1487597678.949621467]: Field 15 of received data is not equal to 0 (86). Unexpected data, ignoring scan
[ WARN] [1487597679.017621456]: Field 15 of received data is not equal to 0 (A). Unexpected data, ignoring scan
[ WARN] [1487597679.085621445]: Field 15 of received data is not equal to 0 (CD). Unexpected data, ignoring scan
[ WARN] [1487597679.153621434]: Field 15 of received data is not equal to 0 (C0). Unexpected data, ignoring scan
[ WARN] [1487597679.217621423]: Field 15 of received data is not equal to 0 (9A). Unexpected data, ignoring scan
[ WARN] [1487597679.285621412]: Field 15 of received data is not equal to 0 (76). Unexpected data, ignoring scan
[ WARN] [1487597679.349621402]: Field 15 of received data is not equal to 0 (19). Unexpected data, ignoring scan
[ WARN] [1487597679.417621391]: Field 15 of received data is not equal to 0 (12). Unexpected data, ignoring scan
[ WARN] [1487597679.485621380]: Field 15 of received data is not equal to 0 (78). Unexpected data, ignoring scan
```

As you can see, it appears that the front LIDAR (TIM561) is mixed up with the rear one (TIM510).

* When I unplug the rear one, the front one works fine (will read without errors)
* When I plug both in, the rear one ` roslaunch sick_tim sick_tim310_1130000m01.launch` works fine but the front won't read anything
* If I launch `roslaunch sick_tim sick_tim551_2050001.launch` I get no data at all
* From what I can tell, regardless of the roslaunch param, they are both recognized as the same model.
* Trying to run two nodes, one for each fails, as the LIBUSB interface can't be acquired.

Can you please help?

mintar, 2017-02-20:
If you launch two nodes, both will try to connect to the first USB device (`device_number = 0`), which will fail for the second one because the LIBUSB interface has already been acquired by the first one. You can avoid this by setting the `device_number` parameter: https://github.com/uos/sick_tim/blob/5c9346a4243a6e43ad004847c80a9fb36574e9ab/launch/sick_tim551_2050001.launch#L17

So start the `sick_tim310` node with the parameter `device_number = 0` and the `sick_551` node with `device_number = 1`, or the other way around (you'll have to try).

It would be cool to have a better way to assign the proper nodes, e.g. by device serial number. I'm happy to accept pull requests! :)

alexge233, 2017-02-20:
Hi mintar, thanks for the help!
Can't believe it was that simple. How is the device number generated?
Again, many thanks!

mintar, 2017-02-21:
Glad to help! The device number *n* simply means "the *n*th device in the list returned by libusb" (AFAIK, ordered first by USB bus, then USB device). It remains stable (also during reboots) as long as you always plug in the devices into the same USB ports and don't switch them around.

closed 2017-02-21 by mintar

Darkproduct, 2021-11-11:
Just to document this here: Libusb does not guarantee consistent device ordering

See https://github.com/libusb/libusb/issues/311:
>If I call libusb_get_device_list with the same context, will I always get the devices in the same order? This works under linux, but is it guaranteed to work on all platforms? (assuming there's no hardware changes between calls).

> No, there is no guarantee of ordering. I'll close this since it doesn't appear to be a bug report. Let me know otherwise.

mintar, 2021-11-11:
Thanks for the information! However, the text you quoted says "This works under Linux". I'd interpret this as saying that consistent ordering isn't guaranteed across platforms, but the Linux implementation just happens to return a consistent ordering. Since probably 99% of the people running our driver use Linux, that could be good enough?

I agree that it's a brittle way of addressing the devices, and it would be better to use something like the device serial number (something that many cameras support), but I don't know if that's supported by the SICK scanners. Note that this problem only happens if you're using the USB interface. On Ethernet, the scanners are properly addressed via their IP address.

Darkproduct, 2021-11-11:
Could be. The `device_number` parameter is not that stable, because this can change by reorganizing all all USB cables. A selection from the serial number would be better. 

But since ROS is going EOL and we now have the official drivers from SICK, I guess this package will go EOL as well. Unfortunately, `sick_scan2` has no interest in supporting USB https://github.com/SICKAG/sick_scan2/issues/22. That's the only reason we currently use this driver with the `ros_bridge`.