uos/sick_tim #46: Symbol lookup error when running from 0.0.10 binary

bennilson, 2017-03-13:
I'm trying to run this package installed via apt-get (ros-indigo-sick-tim 0.0.10-0trusty-20170107-080548-0800). However, when i try running a launch file i get:
/opt/ros/indigo/lib/sick_tim/sick_tim551_2050001: symbol lookup error: /opt/ros/indigo/lib/sick_tim/sick_tim551_2050001: undefined symbol: _ZN8sick_tim13SickTimCommon13rebootScannerEv

I can download the source, build it, and it runs fine. This seems to point to an issue with the way it installs. Has anyone seen this issue or know what is causing it?

I'm running ROS Indigo on Ubuntu 14.04. Thanks!

mintar, 2017-03-14:
I have just checked it and cannot reproduce the bug. For me, it works fine using the exact same version you are using. I suspect your `LD_LIBRARY_PATH` is not set correctly.

mintar, 2017-03-14:
I have just checked it and cannot reproduce the bug. For me, it works fine using the exact same version you are using. I suspect your `LD_LIBRARY_PATH` is not set correctly. This is how mine looks:

```bash
$ echo $LD_LIBRARY_PATH 
/opt/ros/indigo/lib/x86_64-linux-gnu:/opt/ros/indigo/lib
```

bennilson, 2017-03-14:
You are correct. My LD_LIBRARY_PATH was going through another workspace (which had a source copy), set in my bashrc, that I was overriding. After I removed the other workspace sourcing from my bashrc, my LD_LIBRARY_PATH looked like yours and then it worked correctly. 

Thanks for your help with this.

closed 2017-03-14 by bennilson

mintar, 2017-03-14:
Glad to have helped! :)