uos/sick_tim #47: SICK TIM571-2050101 drops connection and does not reconnect

stwirth, 2017-03-21:
This happens to us from time to time using a SICK TIM571-2050101 with the `sick_tim551_2050001` node:
```
[ERROR] [/base_laser] [1489864434.459468694]: sendSOPASCommand: failed attempt to read from socket: 125: Operation canceled
[ERROR] [/base_laser] [1489864434.459550875]: get_datagram: no data available for read after 1000 ms
[ INFO] [/base_laser] [1489864435.459691127]: Failure - attempting to reconnect
[ INFO] [/base_laser] [1489864435.459795559]: Waiting 5 seconds for device to connect.
[ INFO] [/base_laser] [1489864435.460289137]: Succesfully connected to 10.2.98.50:2112
[ERROR] [/base_laser] [1489864436.460389179]: sendSOPASCommand: failed attempt to read from socket: 125: Operation canceled
[ERROR] [/base_laser] [1489864436.460447905]: sendSOPASCommand: no full reply available for read after 1s
[ERROR] [/base_laser] [1489864436.460482355]: SOPAS - Error reading variable 'DeviceIdent'.
[ERROR] [/base_laser] [1489864436.460515051]: sendSOPASCommand: socket not open
[ERROR] [/base_laser] [1489864436.460537546]: SOPAS - Error reading variable 'SerialNumber'.
[ERROR] [/base_laser] [1489864436.460559206]: sendSOPASCommand: socket not open
[ERROR] [/base_laser] [1489864436.460577109]: SOPAS - Error reading variable 'FirmwareVersion'.
[ERROR] [/base_laser] [1489864436.460595450]: sendSOPASCommand: socket not open
[ERROR] [/base_laser] [1489864436.460612651]: SOPAS - Error starting to stream 'LMDscandata'.
[FATAL] [/base_laser] [1489864436.460631811]: Failed to init scanner: 1
[ERROR] [/base_laser] [1489864436.460645745]: Read Error when getting datagram: 1.
[ERROR] [/base_laser] [1489864436.460663384]: sendSOPASCommand: socket not open

SOPAS - Error stopping streaming scan data!
sick_tim driver exiting.
[ INFO] [/base_laser] [1489864436.484668416]: Waiting 5 seconds for device to connect.
[ INFO] [/base_laser] [1489864436.485854342]: Succesfully connected to 10.2.98.50:2112
[ERROR] [/base_laser] [1489864437.486015098]: sendSOPASCommand: failed attempt to read from socket: 125: Operation canceled
[ERROR] [/base_laser] [1489864437.486135199]: sendSOPASCommand: no full reply available for read after 1s
[ERROR] [/base_laser] [1489864437.486190036]: SOPAS - Error reading variable 'DeviceIdent'.
[ERROR] [/base_laser] [1489864437.486240601]: sendSOPASCommand: socket not open
[ERROR] [/base_laser] [1489864437.486278062]: SOPAS - Error reading variable 'SerialNumber'.
[ERROR] [/base_laser] [1489864437.486319564]: sendSOPASCommand: socket not open
[ERROR] [/base_laser] [1489864437.486361547]: SOPAS - Error reading variable 'FirmwareVersion'.
[ERROR] [/base_laser] [1489864437.486399070]: sendSOPASCommand: socket not open
[ERROR] [/base_laser] [1489864437.486431723]: SOPAS - Error starting to stream 'LMDscandata'.
[FATAL] [/base_laser] [1489864437.486464858]: Failed to init scanner: 1
[ERROR] [/base_laser] [1489864437.486491480]: sendSOPASCommand: socket not open

SOPAS - Error stopping streaming scan data!
sick_tim driver exiting.
```
The laser comes back up after a full power cycle of laser & host computer.
This looks like a networking issue to me but I don't know if it is on the laser side or on the host computer.
Do you have any idea on how to debug this further? Unfortunately the issue is not reproducible and has a very low occurrence frequency.

mintar, 2017-03-21:
Sorry, I don't have any idea either. I haven't seen this bug yet, and since as you said it's not reproducible this doesn't sound like a fun bug to fix...

Have you ever tried *only* power-cycling the laser *or* the host?

SICK-NorthAmerica, 2017-03-21:
In addition to @mintar's question, I'd also be curious what firmware revision is running on the TiM571.

stwirth, 2017-03-22:
@mintar @SICK-NorthAmerica thanks for your comments!
@mintar I haven't tried powercycling the laser only. Will do if it happens to occur on a robot that I have physical access to. A simple soft reboot of the host doesn't help.
@SICK-NorthAmerica I see that the firmware version is read in the driver but not used anywhere: https://github.com/uos/sick_tim/blob/indigo/src/sick_tim_common.cpp#L196-L202
Is that information already included in the DeviceIdent string? Here's what I get from diagnostics_agg:
```
 hardware_id: sRA 0 6 TiM571 E V2.45-12.03.15 sRA SerialNumber 8 15170151
```

mintar, 2017-03-22:
Yes, the string includes the firmware version: `V2.45-12.03.15`.

mintar, 2017-03-22:
Just for reference: Our scanner has firmware `V2.51-21.08.15` . I haven't personally seen the problem, but that doesn't mean I'm positive it's not there. I guess many people don't even report a problem that's not happening often, so I might not be aware of it.

@SICK-NorthAmerica : is there a changelog what has been fixed in which firmware version?

akrv, 2018-05-29:
hi,
@mintar and @stwirth 
i have the same problem. 
trying to find a solution for three hours. read the comment about power cycling.
I am ashamed to say but it worked ;) 

is there any solution to this?

steps to reproduce:
I have PC on board the robot which connects over Ethernet to the TIM571.
just reboot the PC without power cycling the whole robot a few time.
you will see that the laser scanner doesn't show up anymore and starting it by had gives the following error as seen in the picture.

[image]

Drashtigp, 2019-09-13:
I had the same issue. But In SOPAS ET, Change the communication protocol  to cola-A ( ASCII ) and save these changes, It works!

Achllle, 2020-05-18:
Ran into the same problem with TIM551. Power cycling did the trick, but would love to see a fix for this.

JiyongOh, 2020-06-25:
I had the same issue as one of **akrv**.
Following the procedures that **Drashtigp** mentioned, I can finally make SICK TiM 571 work on Ubuntu 18.04 and ROS melodic.
Thanks to Drashtigp!

PopovVasil, 2020-12-21:
This happened to me every time for two days. Im using a SICK TIM551-2050101 with the sick_tim551_2050001 node, only USB connection:

[image]

closed 2024-11-27 by mintar