uos/sick_tim #53: MRS1000 - Ubuntu 16.4 - ROS kinetic

VincentTH, 2017-11-17:
Dear all
First my MRS1000 work well on Windows with the tools provide by Sick
But...
I try to use one MRS1000 with ROS kinetic (Ubuntu 16.4)
I well config the Ethernet like propose in **https://github.com/uos/sick_tim/issues/34**
I'm able to ping my lidar on 169.254.80.200 (2112 / 255.255.255.0 / 0.0.0.0)
ping 169.254.80.200
PING 169.254.80.200 (169.254.80.200) 56(84) bytes of data.
64 bytes from 169.254.80.200: icmp_seq=1 ttl=64 time=0.071 ms
64 bytes from 169.254.80.200: icmp_seq=2 ttl=64 time=0.077 ms
64 bytes from 169.254.80.200: icmp_seq=3 ttl=64 time=0.084 ms_

in the 
sick_mrs_1000.launch
<param name="hostname" type="string" value="169.254.80.200"/>
<param name="port" type="string" value="2112"/>

but when I try to launch the ROS node, I have :
### roslaunch sick_tim sick_mrs_1000.launch
... logging to /home/th01_up2/.ros/log/c4817d68-cb97-11e7-b5ef-000732496497/roslaunch-th01-30311.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://th01:35931/
SUMMARY
========
PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /sick_mrs_1000/hostname: 169.254.80.200
 * /sick_mrs_1000/port: 2112
 * /sick_mrs_1000/range_max: 25.0
NODES
  /
    sick_mrs_1000 (sick_tim/sick_mrs_1000)
ROS_MASTER_URI=http://localhost:11311
core service [/rosout] found
process[sick_mrs_1000-1]: started with pid [30329]
[ INFO] [1510927783.024197611]: Waiting 5 seconds for device to connect.
[ERROR] [1510927783.024648276]: Failed to connect to 169.254.80.200:2112
[FATAL] [1510927783.024843050]: Could not connect to host 169.254.80.200:2112
[FATAL] [1510927783.024983489]: Failed to init device: 1
[ERROR] [1510927783.025239108]: write error for command: sEN LMDscandata 0
SOPAS - Error stopping streaming scan data!
sick_tim driver exiting.

Any idea ? What I should change to take hand on my MRS1000 with ROS / Ubuntu
Thank you in advance
Vincent

jspricke, 2017-11-17:
This driver currently does not support the MRS 1000, afaik @spuetz is working on it.

mintar, 2017-11-17:
Yes, I wonder where you found the sick_mrs_1000.launch file...

spuetz, 2017-11-17:
Probably here: https://github.com/mikeferguson/sick_tim/tree/mrs1000
This also does not work with the newer version, because you hove to switch the mode.
I'm currently working on a driver, which is more general. I'll publish a first quick version at the weekend or at the beginning of the next week. 

Regarding your problem above: It looks like your network config is wrong.

VincentTH, 2017-11-17:
Yes I use the MRS1000.launch @ the https://github.com/mikeferguson/sick_tim/tree/mrs1000
Where you publish the new version of the MRS1000 node ? 
and for you, where you see an error in my network config ? a clue ?
Thank you in advance

spuetz, 2017-11-17:
Is your system running in the same ip range? Maybe try first with the standard ip config 192.168.0.1 and your computer 192.168.0.2 in a direct 2p2 connection.
The repo will be on my github page first.

spuetz, 2017-11-29:
Have a look here: https://github.com/spuetz/sick_tim/tree/mrs1000 - a first unstable version. Please tell me if something is not working or if you have questions. Also now the coordinate transformation is right.

closed 2018-01-30 by spuetz

spuetz, 2018-02-01:
Please see https://github.com/uos/sick_tim The MRS1000 driver is now included.