uos/sick_tim #58: libusb.h: No such file or directory

rdockterjr, 2018-02-14:
Not sure if this is specific to my setup or not but thought I would bring it up.

When I clone the sick_tim repo, kinetic branch, I cannot catkin_make the package without modifying header files first.

I initially tried to build the package and received the error libusb.h: No such file or directory. To solve this I had to install the libusb-1.0-0-dev package. However this package installs libusb.h into a sub directory aka /usr/include/libusb-1.0/

Then when I try to rebuild I get errors for sick_tim_common_usb.h and sick_tim_common_mockup.h.

I had to edit these files to include the following:
#include <libusb-1.0/libusb.h>

Again this may be specific to my setup which is
Ubuntu 16.04
ROS Kinetic
Jetson TX1 Arm Processor

jspricke, 2018-02-14:
This sounds like there is something broken with the CMakeLists.txt. It should make sure libusb is there and add the paths as well, here: https://github.com/uos/sick_tim/blob/lunar/CMakeLists.txt#L13
Can you send the complete log of your catkin_make?

rdockterjr, 2018-02-14:
Here is the most recent catkin_make before I implemented my #include change

ubuntu@tegra-ubuntu:~/catkin_ws$ catkin_make
Base path: /home/ubuntu/catkin_ws
Source space: /home/ubuntu/catkin_ws/src
Build space: /home/ubuntu/catkin_ws/build
Devel space: /home/ubuntu/catkin_ws/devel
Install space: /home/ubuntu/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ubuntu/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/ubuntu/catkin_ws/build"
####
[  0%] Built target std_msgs_generate_messages_lisp
[  1%] Built target serial
[  1%] Built target nav_msgs_generate_messages_lisp
[  1%] Built target _novatel_msgs_generate_messages_check_deps_CommonFooter
[  1%] Built target geometry_msgs_generate_messages_lisp
[  1%] Built target _novatel_msgs_generate_messages_check_deps_INSCOV
[  1%] Built target _novatel_msgs_generate_messages_check_deps_Ack
[  1%] Built target _novatel_msgs_generate_messages_check_deps_CommonHeader
[  1%] Built target _novatel_msgs_generate_messages_check_deps_INSPVAX
[  1%] Built target _novatel_msgs_generate_messages_check_deps_CORRIMUDATA
[  1%] Built target std_msgs_generate_messages_py
[  1%] Built target geometry_msgs_generate_messages_py
[  1%] Built target _novatel_msgs_generate_messages_check_deps_BESTPOS
[  1%] Built target nav_msgs_generate_messages_nodejs
[  1%] Built target nav_msgs_generate_messages_py
[  1%] Built target std_msgs_generate_messages_nodejs
[  1%] Built target geometry_msgs_generate_messages_nodejs
[  1%] Built target std_msgs_generate_messages_cpp
[  1%] Built target std_msgs_generate_messages_eus
[  1%] Built target nav_msgs_generate_messages_cpp
[  1%] Built target geometry_msgs_generate_messages_cpp
[  1%] Built target nav_msgs_generate_messages_eus
[  1%] Built target geometry_msgs_generate_messages_eus
[  1%] Built target delay_loader_raise
[  1%] Built target angle_to_loader_cmd
[  2%] Built target height_to_loader_cmd
[  2%] Built target razor_imu_9dof_gencfg
[  2%] Built target diagnostic_msgs_generate_messages_py
[  2%] Built target roscpp_generate_messages_lisp
[  6%] Built target sbpl
.
.
.
Scanning dependencies of target robot_localization_generate_messages_cpp
[ 42%] Generating C++ code from robot_localization/SetDatum.srv
[ 43%] Generating C++ code from robot_localization/SetPose.srv
[ 43%] Built target robot_localization_generate_messages_cpp
Scanning dependencies of target robot_localization_generate_messages_lisp
[ 43%] Generating Lisp code from robot_localization/SetDatum.srv
[ 44%] Generating Lisp code from robot_localization/SetPose.srv
[ 44%] Built target robot_localization_generate_messages_lisp
Scanning dependencies of target robot_localization_generate_messages_py
[ 44%] Generating Python code from SRV robot_localization/SetDatum
[ 45%] Generating Python code from SRV robot_localization/SetPose
[ 45%] Generating Python srv __init__.py for robot_localization
[ 45%] Built target robot_localization_generate_messages_py
Scanning dependencies of target robot_localization_generate_messages_eus
[ 45%] Generating EusLisp code from robot_localization/SetDatum.srv
[ 45%] Generating EusLisp code from robot_localization/SetPose.srv
[ 45%] Generating EusLisp manifest code for robot_localization
[ 45%] Built target robot_localization_generate_messages_eus
[ 45%] Linking CXX shared library /home/ubuntu/catkin_ws/devel/lib/libpose_follower.so
[ 45%] Built target pose_follower
[ 45%] Building CXX object Northstar/sick_tim/CMakeFiles/sick_tim_3xx.dir/src/sick_tim_common_usb.cpp.o
In file included from /home/ubuntu/catkin_ws/src/Northstar/sick_tim/src/sick_tim_common_usb.cpp:39:0:
/home/ubuntu/catkin_ws/src/Northstar/sick_tim/include/sick_tim/sick_tim_common_usb.h:45:20: fatal error: libusb.h: No such file or directory
compilation terminated.
Northstar/sick_tim/CMakeFiles/sick_tim_3xx.dir/build.make:86: recipe for target 'Northstar/sick_tim/CMakeFiles/sick_tim_3xx.dir/src/sick_tim_common_usb.cpp.o' failed
make[2]: *** [Northstar/sick_tim/CMakeFiles/sick_tim_3xx.dir/src/sick_tim_common_usb.cpp.o] Error 1
CMakeFiles/Makefile2:4641: recipe for target 'Northstar/sick_tim/CMakeFiles/sick_tim_3xx.dir/all' failed
make[1]: *** [Northstar/sick_tim/CMakeFiles/sick_tim_3xx.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 45%] Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/web_machine_control/listener
[ 45%] Built target listener
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

jspricke, 2018-02-14:
Can you do that again with cleaning up the build space first? I.e.:
``
rm -r build devel
catkin_make
``
And send the complete output.

rdockterjr, 2018-02-14:
jsprickle,

Well now I feel dumb. After deleting build and devel, the catkin_make succeeded. I must have just been stuck in a weird build state after installing libusb.

Thanks for the help. I'll mark this as closed.

-Rod

closed 2018-02-14 by rdockterjr