uos/sick_tim #62: Sick_tim sick_mrs1000: Edit in launch file is not working

danzijl, 2018-04-25:
I was trying to connect the MRS1000 with ubuntu 16.04.  
I edited the launch file to my scanners ip-adres and port. (169.254.146.33) (2112)
[image]

I also added a network as can be seen:
[image]

when trying to run the launch file i get the following error:

`dvzijl@dvzijl-VirtualBox:~$ roslaunch sick_tim sick_mrs1000.launch
... logging to /home/dvzijl/.ros/log/a1a92df0-487d-11e8-a03d-080027e11e98/roslaunch-dvzijl-VirtualBox-8584.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dvzijl-VirtualBox:34491/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /sick_mrs1000/hostname: 192.168.0.1
 * /sick_mrs1000/port: 2111

NODES
  /
    sick_mrs1000 (sick_tim/sick_mrs1000)

auto-starting new master
process[master]: started with pid [8594]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a1a92df0-487d-11e8-a03d-080027e11e98
process[rosout-1]: started with pid [8607]
started core service [/rosout]
process[sick_mrs1000-2]: started with pid [8617]
[ INFO] [1524656505.957309267]: Waiting 5 seconds for device to connect.
[ERROR] [1524656505.958788048]: Failed to connect to 192.168.0.1:2111
[FATAL] [1524656505.958812768]: Could not connect to host 192.168.0.1:2111
[FATAL] [1524656505.958830097]: Failed to init device: 1
[ERROR] [1524656505.959843522]: write error for command: sEN LMDscandata 0

SOPAS - Error stopping streaming scan data!
sick_tim driver exiting.
[ INFO] [1524656505.984143105]: Waiting 5 seconds for device to connect.
[ERROR] [1524656505.985229911]: Failed to connect to 192.168.0.1:2111
[FATAL] [1524656505.985249588]: Could not connect to host 192.168.0.1:2111
[FATAL] [1524656505.985288035]: Failed to init device: 1
[ERROR] [1524656505.985955430]: write error for command: sEN LMDscandata 0`

While i edited the launch file i still get the error with it trying to connect to 192.168.0.1:2111?

jspricke, 2018-04-26:
I would propose to install Ubuntu without virtualization.

mintar, 2018-07-12:
Closing due to inactivity.

closed 2018-07-12 by mintar