uos/sick_tim #64: SICK MRS1000 and multiple echoes

ssheikki, 2018-08-07:
The sick_tim ROS driver seems to work well when the SICK MRS1000 is sending only first or last echo: both range and intensity measurements are published correctly. If the MRS1000 is configured to send all the three echoes (with or without intensities) the sick_tim prints repeatedly following warnings:
[ WARN] [1533648107.591137655]: Number of RSSI data is lower than number of range data (1101 vs 5061
[ WARN] [1533648107.611167093]: Number of RSSI data is lower than number of range data (1101 vs 4944
and no data is published. 

So it seems currently there is no multiple echoes (three in case of SICK MRS1000) support in the sick_tim ROS driver, but it would be great if it could be added!

mintar, 2018-08-08:
@spuetz , could you have a look?

Master-Matt, 2018-08-31:
I can confirm. Had the same issue just yesterday. I'd like to use all echos.

mintar, 2024-11-27:
Won't fix, use [SICKAG/sick_scan_xd](https://github.com/SICKAG/sick_scan_xd).

closed 2024-11-27 by mintar