uos/sick_tim #70: Index signal (15 Hz), corresponds to measurement at 90°？

maxvalor, 2018-09-15:
The title was copied from Operating_instructions_TiM55x_TiM56x_TiM57x_en_IM0053143.PDF in the table "Function of the synchronization output (SYNC/De-
vice Ready)". And the question which makes me confused is that if it is correct for the Tim561?Can it be configured?

mintar, 2018-09-18:
**TL;DR:** You don't need the hardware sync in order to run the sick_tim driver.

**Long version:**

No, the sick_tim driver does not yet support hardware sync. The reason is that the driver currently only reads from USB or Ethernet, and the hardware sync is a digital output (i.e. a line that has to be connected directly to the processing computer).

The advantage of doing this would be that you would get more precise time stamps for when a laser scan was recorded. Without the hardware sync, there is always a tiny imprecision because of the Ethernet / USB latency and jitter.

How to do it?

* Wire the hardware sync cable to a digital I/O module that's connected to the PC. You should probably **not** use a USB module (because of USB latency), but rather something like RS232 (but your PC needs a RS232 port of course).

* Write the software to detect the sync signal from the digital I/O board.

* Use the time stamp from the hardware sync in the published laser scans from sick_tim.

This seems a lot of hassle, and I've never bothered to do this because the tiny imprecisions caused by USB / Ethernet latency were never a problem for my use cases.

maxvalor, 2018-09-20:
Got it. Thanks.

closed 2018-09-20 by maxvalor