uos/sick_tim #72: Less fields than expected for 811 data points (1660). Ignoring scan

chennuo0125, 2018-11-08:
thanks you for your package. i have a sick tim571,but when i " roslaunch sick_tim sick_tim571_2050101.launch" , i got follow problem :

[ INFO] [1541644593.482460014]: Waiting 5 seconds for device to connect.
[ INFO] [1541644593.482886536]: Succesfully connected to 169.254.66.113:2112
[ WARN] [1541644593.551938275]: Less fields than expected for 811 data points (1660). Ignoring scan
[ WARN] [1541644593.618800364]: Less fields than expected for 811 data points (1660). Ignoring scan
[ WARN] [1541644593.685426119]: Less fields than expected for 811 data points (1660). Ignoring scan
[ WARN] [1541644593.752336910]: Less fields than expected for 811 data points (1660). Ignoring scan
[ WARN] [1541644593.818296715]: Less fields than expected for 811 data points (1660). Ignoring scan
[ WARN] [1541644593.885475636]: Less fields than expected for 811 data points (1660). Ignoring scan
[ WARN] [1541644593.951638890]: Less fields than expected for 811 data points (1660). Ignoring scan
[ WARN] [1541644594.018524931]: Less fields than expected for 811 data points (1660). Ignoring scan
[ WARN] [1541644594.084968651]: Less fields than expected for 811 data points (1660). Ignoring scan

as you can see ,internet connected successfully, so hardware is ok. could you tell me how can i process it?  thanks !
my computer system is ubuntu16.04, ros kinetic.

chennuo0125, 2018-11-08:
thank your reply , i have record datagram and put it in the link below:
https://github.com/chennuo0125/sick_tim/tree/melodic/data
my launch file is :
<?xml version="1.0"?>
<launch>
  <param name="robot_description" command="$(find xacro)/xacro --inorder '$(find sick_tim)/urdf/example.urdf.xacro'" />
  <node name="robot_state_publisher" pkg="robot_state_publisher" type="state_publisher" />

  <node name="sick_tim571_2050101" pkg="sick_tim" type="sick_tim551_2050001" respawn="false" output="screen">
    
      <param name="min_ang" type="double" value="-2.35619449019" />
      <param name="max_ang" type="double" value="2.35619449019" />
      <param name="intensity" type="bool" value="True" />
      <param name="skip" type="int" value="0" />
      <param name="frame_id" type="str" value="laser" />
      <param name="time_offset" type="double" value="-0.001" />
      <param name="publish_datagram" type="bool" value="true" />
      <param name="subscribe_datagram" type="bool" value="false" />
      <param name="device_number" type="int" value="0" />
      <param name="range_min" type="double" value="0.05" />
    <param name="range_max" type="double" value="25.0" />

    
    

    
         <param name="hostname" type="string" value="169.254.66.113" />
         <param name="port" type="string" value="2112" />
         <param name="timelimit" type="int" value="5" />
  </node>
</launch>

chennuo0125, 2018-11-08:
thank your reply , i have record datagram and put it in the link below:
https://github.com/chennuo0125/sick_tim/tree/melodic/data
my launch file is :
<launch>
  <param name="robot_description" command="$(find xacro)/xacro --inorder '$(find sick_tim)/urdf/example.urdf.xacro'" />
  <node name="robot_state_publisher" pkg="robot_state_publisher" type="state_publisher" />
  <node name="sick_tim571_2050101" pkg="sick_tim" type="sick_tim551_2050001" respawn="false" output="screen">
    
      <param name="min_ang" type="double" value="-2.35619449019" />
      <param name="max_ang" type="double" value="2.35619449019" />
      <param name="intensity" type="bool" value="True" />
      <param name="skip" type="int" value="0" />
      <param name="frame_id" type="str" value="laser" />
      <param name="time_offset" type="double" value="-0.001" />
      <param name="publish_datagram" type="bool" value="true" />
      <param name="subscribe_datagram" type="bool" value="false" />
      <param name="device_number" type="int" value="0" />
      <param name="range_min" type="double" value="0.05" />
    <param name="range_max" type="double" value="25.0" />

    
    

    
         <param name="hostname" type="string" value="169.254.66.113" />
         <param name="port" type="string" value="2112" />
         <param name="timelimit" type="int" value="5" />
  </node>
</launch>

chennuo0125, 2018-11-08:
thank your reply , i have record datagram and put it in the link below:
https://github.com/chennuo0125/sick_tim/tree/melodic/data

mintar, 2018-11-08:
Your scanner has the device string set to the empty string. By default, it is set to the string "not defined". There was a bug in the parser that caused this issue. I have just pushed a commit that fixes the problem. To use it, you'll have to pull the latest code and build from source.

Thanks for your bug report!

closed 2018-11-08 by mintar

chennuo0125, 2018-11-08:
@mintar  what can i do to process it ? i still encounter the problem.
my datagram have been put in the link below : 
https://github.com/chennuo0125/sick_tim/tree/melodic/data

[ INFO] [1541644593.482460014]: Waiting 5 seconds for device to connect.
[ INFO] [1541644593.482886536]: Succesfully connected to 169.254.66.113:2112
[ WARN] [1541644593.551938275]: Less fields than expected for 811 data points (1660). Ignoring scan
[ WARN] [1541644593.618800364]: Less fields than expected for 811 data points (1660). Ignoring scan
[ WARN] [1541644593.685426119]: Less fields than expected for 811 data points (1660). Ignoring scan
[ WARN] [1541644593.752336910]: Less fields than expected for 811 data points (1660). Ignoring scan
[ WARN] [1541644593.818296715]: Less fields than expected for 811 data points (1660). Ignoring scan
[ WARN] [1541644593.885475636]: Less fields than expected for 811 data points (1660). Ignoring scan
[ WARN] [1541644593.951638890]: Less fields than expected for 811 data points (1660). Ignoring scan
[ WARN] [1541644594.018524931]: Less fields than expected for 811 data points (1660). Ignoring scan
[ WARN] [1541644594.084968651]: Less fields than expected for 811 data points (1660). Ignoring scan

mintar, 2018-11-08:
You should pull the latest code and compile it from source.

chennuo0125, 2018-11-09:
thank you , the problem has been solved !

commit 995cde8f (2018-11-08): TiM551/TiM571: Fix HEADER_FIELDS calculation

When the device name is not set, the number of footer fields is 5, not 6
as assumed before. This brings the total number of header + footer
fields plus the field that indicates RSSI data to 32, not 33.

Fixes #72.