uos/sick_tim #76: TIM 561 launch file throwing runtime exception

Megh-Thakkar, 2019-03-26:
I am new to ROS and am
[image]
 using a sick TIM 561 scanner with a USB on a linux system. While launching the scanner using 
```bash
$ roslaunch sick_tim sick_tim551_2050001.launch
```
it is throwing an exception (screenshot attached). Please help me with the same.

mintar, 2019-03-26:
You probably have the parameter `use_sim_time` set to `true`. To fix this, run:

```bash
rosparam set use_sim_time false
```

... and then restart the launch file.

mintar, 2019-03-26:
I've added a commit that should avoid the runtime error and print a more helpful warning.

closed 2019-03-26 by mintar

Megh-Thakkar, 2019-03-26:
Thanks a lot for your prompt response. I will definitely get back to you in case of further queries.