uos/sick_tim #77: connect sick_tim561 to pc through USB problem

Lip0, 2019-05-06:
hi, I want connect sick_tim561 lidar to pc using  package sick_tim. When I connect lidar through Ethernet, it can be successfully to read lidar data. But when I connect lidar through USB, it appears an error. I post it below.
[image]

jspricke, 2019-05-06:
Did you install sick_tim from source or the the ROS package repository?
Did you install the udev rules: https://github.com/uos/sick_tim/issues/77
What's the output of `lsusb` and of `ls -l /dev/bus/*`

Lip0, 2019-05-06:
yes,I install the udev rules,  current user is a member of the plugdev group .
 the output of lsusb is the type of USB cable,the led of lidar is green.
[image]
[image]

Lip0, 2019-05-06:
> Did you install sick_tim from source or the the ROS package repository?
> Did you install the udev rules: #77
> What's the output of `lsusb` and of `ls -l /dev/bus/*`
see below,thx

jspricke, 2019-05-06:
The ls output is missing the subdirs, can you try `ls -lR /dev/bus/usb` instead?

Lip0, 2019-05-07:
> ls -lR /dev/bus/usb
[image]
[image]

jspricke, 2019-05-07:
The lsusb output implies that you want to connect to the device through a third party USB to serial converter. This driver only supports direct connection to the USB plug of the device or Ethernet.

Lip0, 2019-05-07:
> The lsusb output implies that you want to connect to the device through a third party USB to serial converter. This driver only supports direct connection to the USB plug of the device or Ethernet.

you mean the usb cable have problem? I need to change a usb cable?

jspricke, 2019-05-07:
For USB you need to connect to the USB port of the laser scanner, which is behind the USB logo on the left on this picture: https://cdn.sick.com/media/ZOOM/5/25/025/IM0051025.png.

Closing because it's not a driver problem.

closed 2019-05-07 by jspricke