uos/sick_tim #78: terminate called after throwing an instance of 'std::runtime_error'

shanpenghui, 2019-06-21:
After I run 
```
sudo apt upgrade
```
On my computer which is Ubuntu18.04 ,ROS kinetic,i can not launch the sick_tim.
```
roslaunch sick_tim demo.launch
```
The error tips:
```
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.102:33235/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /sick_tim571_2050101/frame_id: laser_frame
 * /sick_tim571_2050101/hostname: 192.168.1.112
 * /sick_tim571_2050101/intensity: False
 * /sick_tim571_2050101/max_ang: 1.47
 * /sick_tim571_2050101/min_ang: -1.47
 * /sick_tim571_2050101/port: 2112
 * /sick_tim571_2050101/range_max: 10.0
 * /sick_tim571_2050101/timelimit: 5

NODES
  /
    base_link_to_laser5 (tf/static_transform_publisher)
    sick_tim571_2050101 (sick_tim/sick_tim551_2050001)

ROS_MASTER_URI=http://localhost:11311

process[sick_tim571_2050101-1]: started with pid [24076]
process[base_link_to_laser5-2]: started with pid [24089]
terminate called after throwing an instance of 'std::runtime_error'
  what():  Time is out of dual 32-bit range
[sick_tim571_2050101-1] process has died [pid 24076, exit code -6, cmd /sick_ws/devel/lib/sick_tim/sick_tim551_2050001 scan:=sick_scan __name:=sick_tim571_2050101 __log:=/home//.ros/log/b0b341c6-93c9-11e9-ab34-a43bfa91b9d7/sick_tim571_2050101-1.log].
log file: /.ros/log/b0b341c6-93c9-11e9-ab34-a43bfa91b9d7/sick_tim571_2050101-1*.log
^C[base_link_to_laser5-2] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
I consider if anywhere divide zero but don't where.Anyone can help?

mintar, 2019-06-21:
It's not a divide by zero problem; the runtime error happens when a ros::Time is assigned a negative value, such as in `header.stamp.fromSec(-1.0)`. This used to happen when the ROS parameter use_sim_time was set to 0, which I fixed in e9c7873c. That commit was pushed out during the latest release, so it could somehow be causing your bug.

However, I've tried to reproduce it and couldn't. It works on Kinetic and Melodic, both when built from source and with the latest binary release (0.0.16). So I'm a bit lost what happens here.

All that I can say is that it [works on my machine](https://blog.codinghorror.com/the-works-on-my-machine-certification-program/)!

[image]

Sorry about that. :)

mintar, 2019-06-21:
Oh, and one small remark (not related to your bug): If you're using a SICK TiM571 scanner, you should set `range_max` to `25.0`.

closed 2019-06-27 by mintar

DSaikrishnareddy, 2020-07-28:
> After I run
> 
> ```
> sudo apt upgrade
> ```
> 
> On my computer which is Ubuntu18.04 ,ROS kinetic,i can not launch the sick_tim.
> 
> ```
> roslaunch sick_tim demo.launch
> ```
> 
> The error tips:
> 
> ```
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://192.168.1.102:33235/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /robot_description: <?xml version="1....
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.14
>  * /sick_tim571_2050101/frame_id: laser_frame
>  * /sick_tim571_2050101/hostname: 192.168.1.112
>  * /sick_tim571_2050101/intensity: False
>  * /sick_tim571_2050101/max_ang: 1.47
>  * /sick_tim571_2050101/min_ang: -1.47
>  * /sick_tim571_2050101/port: 2112
>  * /sick_tim571_2050101/range_max: 10.0
>  * /sick_tim571_2050101/timelimit: 5
> 
> NODES
>   /
>     base_link_to_laser5 (tf/static_transform_publisher)
>     sick_tim571_2050101 (sick_tim/sick_tim551_2050001)
> 
> ROS_MASTER_URI=http://localhost:11311
> 
> process[sick_tim571_2050101-1]: started with pid [24076]
> process[base_link_to_laser5-2]: started with pid [24089]
> terminate called after throwing an instance of 'std::runtime_error'
>   what():  Time is out of dual 32-bit range
> [sick_tim571_2050101-1] process has died [pid 24076, exit code -6, cmd /sick_ws/devel/lib/sick_tim/sick_tim551_2050001 scan:=sick_scan __name:=sick_tim571_2050101 __log:=/home//.ros/log/b0b341c6-93c9-11e9-ab34-a43bfa91b9d7/sick_tim571_2050101-1.log].
> log file: /.ros/log/b0b341c6-93c9-11e9-ab34-a43bfa91b9d7/sick_tim571_2050101-1*.log
> ^C[base_link_to_laser5-2] killing on exit
> shutting down processing monitor...
> ... shutting down processing monitor complete
> done
> ```
> 
> All of the demo.launch file is:
> 
> ```
> <?xml version="1.0"?>
> <launch>
>   <param name="robot_description" command="$(find xacro)/xacro.py '$(find sick_tim)/urdf/example.urdf.xacro'" />
> 
>   <node name="sick_tim571_2050101" pkg="sick_tim" type="sick_tim551_2050001" respawn="false" output="screen">
>       <param name="min_ang" type="double" value="-1.47" />
>       <param name="max_ang" type="double" value="1.47" />
>       <param name="intensity" type="bool" value="False" />
>       <param name="frame_id" type="str" value="laser_frame" />
>     
>     
>     <param name="range_max" type="double" value="10.0" />
> 
>     
>     
> 
>     
>    <param name="hostname" type="string" value="192.168.1.112" />
>          <param name="port" type="string" value="2112" />
>          <param name="timelimit" type="int" value="5" />
>   <remap from="scan" to="sick_scan"/>
>   </node>
>   <node pkg="tf" type="static_transform_publisher" name="base_link_to_laser5"
>    args="0.40 0.0 0.4 -3.2 -3.1415926  0.0 /base_footprint /laser_frame 40" />
> </launch>
> ```
> 
> I consider if anywhere divide zero but don't know where. Anyone can help?

I too got the same error but with different launch file.

DSaikrishnareddy, 2020-07-28:
On my computer which is Ubuntu18.04 ,ROS melodic,i can not launch the tiago_opencv_tutorial.
`roslaunch tiago_opencv_tutorial keypoint_tutorial.launch`
The error tips:
`... logging to /home/dell/.ros/log/d0d33860-d0bf-11ea-a40c-54353005d2aa/roslaunch-skdr-4324.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://skdr:45901/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.6

NODES
  /
    gui (tiago_opencv_tutorial/gui_find_keypoints.py)
    vision_code (tiago_opencv_tutorial/find_keypoints)

ROS_MASTER_URI=http://localhost:11311

process[gui-1]: started with pid [4364]
process[vision_code-2]: started with pid [4365]
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unsupported type of detector SURF
[vision_code-2] process has died [pid 4365, exit code -6, cmd /home/dell/tiago_dual_public_ws/devel/lib/tiago_opencv_tutorial/find_keypoints __name:=vision_code __log:=/home/dell/.ros/log/d0d33860-d0bf-11ea-a40c-54353005d2aa/vision_code-2.log].
log file: /home/dell/.ros/log/d0d33860-d0bf-11ea-a40c-54353005d2aa/vision_code-2*.log
^C[gui-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`
All of the keypoint_tutorial.launch file is:
`<?xml version="1.0" encoding="UTF-8"?>
<launch>
 <node name="gui" pkg="tiago_opencv_tutorial" type="gui_find_keypoints.py"/>
 <node name="vision_code" pkg="tiago_opencv_tutorial" type="find_keypoints" output="screen"/>
</launch>`

Can anyone please help me in this.

DSaikrishnareddy, 2020-07-28:
All of the keypoint_tutorial.launch file is:
?xml version="1.0" encoding="UTF-8"?>
<launch>
 <node name="gui" pkg="tiago_opencv_tutorial" type="gui_find_keypoints.py"/>
 <node name="vision_code" pkg="tiago_opencv_tutorial" type="find_keypoints" output="screen"/>
</launch>

Can anyone please help me in this.

mintar, 2020-08-10:
@DSaikrishnareddy : This is the completely wrong repo for your question.