uos/sick_tim #79: SICK TIM651 - Frequency too low

MLouedec, 2019-06-27:
Hello,

I am using a Sick_Tim561 and the sick_tim package for mobile robot localization. But while the scanning frequency is at 15Hz, the topic /scan is only updated at 3Hz. So far I haven't been able to know why. Do you have any idea ?

The sensor is linked to a Raspberry Pi with an Ethernet cable. But the rosmaster is on my laptop which is linked by WiFi to the Raspberry. Moreover I use the launch file sick_tim551_2050001.launch .

Here is a rosbag:
[2019-06-27-09-23-07.bag.tar.gz](https://github.com/uos/sick_tim/files/3333455/2019-06-27-09-23-07.bag.tar.gz)

Have a nice day.

jspricke, 2019-06-27:
Does it give 15hz when connecting the scanner to your computer?
What is the load on CPU, USB and Ethernet on your Pi?

mintar, 2019-06-27:
This is weird. According to the diagnostics messages, the frequency is *exactly* 3.000 Hz, +/- 0.01 Hz. This suggests that it's actually the scanner that is limiting the frequency and not any sort of network or CPU problem. I haven't seen this before; could you open the scanner in SopasET on Windows and poke around in the configuration options? Maybe the frequency there is limited to 3 Hz.

MLouedec, 2019-06-27:
On my computer, the topic also gives 3Hz.
The node only takes 1% of the Raspberry Pi.
The data from Ethernet is received at 150kBit/s.

mintar, 2019-06-27:
What about the SopasET?

MLouedec, 2019-06-27:
I haven't seen anything related to 3Hz in the configuration options. Here are the screenshots of all the configurations :
[SOPAS ET Screenshots.zip](https://github.com/uos/sick_tim/files/3333890/SOPAS.ET.Screenshots.zip)

mintar, 2019-06-27:
That's really weird. Does the "scan view" in Sopas ET update at 3 Hz or 15 Hz?

MLouedec, 2019-06-27:
the "scan view" updates at 3Hz.

mintar, 2019-06-27:
In that case, it doesn't seem to be a problem with the driver. As soon as you get the scanner to work at 15 Hz in Sopas ET, it should also work with our driver. Maybe try getting support from SICK at https://supportportal.sick.com/ .

MLouedec, 2019-06-27:
Ok, thank you for the advice.

SICK-NorthAmerica, 2019-06-27:
Hello, my name is Aaron Rothmeyer and I'm the US product manager for SICK lidar.  I agree that this is probably hardware/configuration related (not driver) but it's an issue I haven't seen before.  Please contact me directly at aaron.rothmeyer@sick.com and I'll help with the next steps of troubleshooting.

MLouedec, 2019-07-05:
I have finally solved the issue, the problem was coming from the TiM651 : it had been reconfigured by frame by one of my colleagues. Thus, after reconfigurating the sensor, the topic frequency is now at 15Hz.
Thank you for you help.

mintar, 2019-07-05:
Thanks for reporting back! Glad that it works for you now.

closed 2019-07-05 by mintar