uos/sick_tim #81: No Intensity available on RVIZ

ROS-lover, 2019-10-03:
Hi there,

I ran the ROS wrapper and noticed that the intensity is only available at the /scan topic which means only one ring (out of) for sick mrs1000. Does that mean the intensity data is available for only one ring?

Thanks!

mintar, 2019-10-04:
@spuetz : Any idea?

ROS-lover, 2019-10-04:
@mintar @spuetz Thanks for the attention. I looked at the source code, it looks like that the rosparam 'intensity' which can be given to the node upon lunching, is never used through out the source code, it looks like this repository is still under development. I found another repository that I used and I can access the intensity data. I can share it here for the reference:  http://wiki.ros.org/sick_scan
Thanks anyways!

mintar, 2019-10-05:
> I looked at the source code, it looks like that the rosparam 'intensity' which can be given to the node upon lunching, is never used through out the source code

Well, look again. The parameter `intensity` is used here:

https://github.com/uos/sick_tim/blob/4abfbe8d1f455859efae9afd66175921e4bf28bd/src/sick_mrs1000_parser.cpp#L286

> it looks like this repository is still under development. I found another repository that I used and I can access the intensity data. I can share it here for the reference: http://wiki.ros.org/sick_scan

Yeah, I know about sick_scan. It's a fork of our repository, and actually we've helped the current authors improve their code a couple of times. It's absolutely not true that this repo is more "under development" than theirs - by now, I'd say their repo has become as mature and stable as ours. sick_scan supports more different scanners than sick_tim though.

Adding intensities to the point cloud is still a reasonable feature request, so I'll leave this issue open and update the title to be more specific.

ROS-lover, 2019-10-06:
Right, my bad, found the usage of intensity rosparam as you mentioned. Thanks!

closed 2024-11-27 by mintar