uos/sick_tim #83: Error while I use roslaunch for tim561

glakshay, 2019-10-14:
My command:
~/catkin_ws/src/sick_tim$ roslaunch sick_tim sick_tim551_2050001.launch model:=urdf/example.urdf.xacro 
Error:
SOPAS - Error stopping streaming scan data!
sick_tim driver exiting.
[ INFO] [1571081574.838301564, 10034.503000000]: Waiting 5 seconds for device to connect.
[ERROR] [1571081574.838361202, 10034.503000000]: Failed to connect to 130.215.223.255:2112
[FATAL] [1571081574.838383549, 10034.503000000]: Could not connect to host 130.215.223.255:2112
[FATAL] [1571081574.838399807, 10034.503000000]: Failed to init device: 1
[ERROR] [1571081574.838445681, 10034.503000000]: write error for command: sEN LMDscandata 0

I have added my whole sick_tim package. I also want it to spawn it in gazebo but i get the error
Error [parser_urdf.cc:3408] Unable to call parseURDF on robot model
Error [parser.cc:340] parse as old deprecated model file failed.

[sick_tim.zip](https://github.com/uos/sick_tim/files/3726127/sick_tim.zip)

Any help will be appreciated. Thanks

mintar, 2019-10-15:
The driver cannot reach the scanner over the network. Make sure that `ping 130.215.223.255` works.

glakshay, 2019-10-19:
Thanks for the response. i tried to ping as you suggested and I get this, 
"--- 130.215.233.255 ping statistics ---
47 packets transmitted, 0 received, 100% packet loss, time 47092ms"
So, nothing is being received I sense? and how do I fix this error?

closed 2020-01-17 by mintar