uos/sick_tim #85: Roslaunch for sick_tim_561

glakshay, 2019-10-19:
I connected the Lidar sensor to my laptop using the USB port and I tried to use the command,
roslaunch sick_tim sick_tim551_2050001.launch

I get the following error even though I am not using TCP and I commented the TCP part in launch file.
[image]
The sick_tim for refernce purpose:
[sick_tim.zip](https://github.com/uos/sick_tim/files/3747007/sick_tim.zip)

glakshay, 2019-10-19:
Also, do we need to install ROS driver from the downloads section of the product's website?

mintar, 2019-10-21:
If you want to use the USB connection, you need to comment out (remove) the following `hostname` parameter from the launch file. There's no need to install anything from the SICK website.

closed 2020-01-17 by mintar