uos/sick_tim #86: SICK TIM310-1030000S01 - LIBUSB - Read Error: -7.

Nepomuk153, 2020-01-14:
Hello, i have problems with the USB connection to Lidar with ROS. 

- connection with usb to SOPAS  works.
- udev rules are created and iam a member of the plugdev group.

PARAMETERS
Ubuntu 16.04.16 
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

Device: TIM310-1030000S01
Software-version:S1.22
[ROS_Info_while_launchfile_started.txt](https://github.com/uos/sick_tim/files/4059022/ROS_Info_while_launchfile_started.txt)

The published messages on topic diagnostics:
`$rostopic echo /diagnostics
header: 
  seq: 1
  stamp: 
    secs: 1579001761
    nsecs: 844727546
  frame_id: ''
status: 
  - 
    level: 2
    name: "sick_tim310s01: /scan topic status"
    message: "LIBUSB - Read Error."
    hardware_id: ''
    values: []
---
header: 
  seq: 2
  stamp: 
    secs: 1579001761
    nsecs: 844962355
  frame_id: ''
status: 
  - 
    level: 2
    name: "sick_tim310s01: /scan topic status"
    message: "SOPAS - Error reading variable 'DeviceIdent'."
    hardware_id: ''
    values: []
---
header: 
  seq: 3
  stamp: 
    secs: 1579001762
    nsecs: 865008806
  frame_id: ''
status: 
  - 
    level: 2
    name: "sick_tim310s01: /scan topic status"
    message: "LIBUSB - Read Error."
    hardware_id: ''
    values: []
---
header: 
  seq: 4
  stamp: 
    secs: 1579001762
    nsecs: 865197905
  frame_id: ''
status: 
  - 
    level: 2
    name: "sick_tim310s01: /scan topic status"
    message: "SOPAS - Error reading variable 'SerialNumber'."
    hardware_id: ''
    values: []
---
header: 
  seq: 5
  stamp: 
    secs: 1579001763
    nsecs: 871177822
  frame_id: ''
status: 
  - 
    level: 2
    name: "sick_tim310s01: /scan topic status"
    message: "LIBUSB - Read Error."
    hardware_id: ''
    values: []
---
header: 
  seq: 6
  stamp: 
    secs: 1579001763
    nsecs: 871285636
  frame_id: ''
status: 
  - 
    level: 2
    name: "sick_tim310s01: /scan topic status"
    message: "SOPAS - Error reading variable 'FirmwareVersion'."
    hardware_id: ''
    values: []
---
header: 
  seq: 7
  stamp: 
    secs: 1579001764
    nsecs: 884093484
  frame_id: ''
status: 
  - 
    level: 2
    name: "sick_tim310s01: /scan topic status"
    message: "LIBUSB - Read Error."
    hardware_id: ''
    values: []
---
header: 
  seq: 8
  stamp: 
    secs: 1579001764
    nsecs: 884344162
  frame_id: ''
status: 
  - 
    level: 2
    name: "sick_tim310s01: /scan topic status"
    message: "SOPAS - Error reading variable 'devicestate'."
    hardware_id: ''
    values: []
---
header: 
  seq: 9
  stamp: 
    secs: 1579001765
    nsecs: 889656554
  frame_id: ''
status: 
  - 
    level: 2
    name: "sick_tim310s01: /scan topic status"
    message: "LIBUSB - Read Error."
    hardware_id: ''
    values: []
---
header: 
  seq: 10
  stamp: 
    secs: 1579001765
    nsecs: 889838481
  frame_id: ''
status: 
  - 
    level: 2
    name: "sick_tim310s01: /scan topic status"
    message: "SOPAS - Error starting to stream 'LMDscandata'."
    hardware_id: ''
    values: []
---
header: 
  seq: 11
  stamp: 
    secs: 1579001766
    nsecs: 900339651
  frame_id: ''
status: 
  - 
    level: 2
    name: "sick_tim310s01: /scan topic status"
    message: "LIBUSB - Read Error."
    hardware_id: ''
    values: []
---
header: 
  seq: 0
  stamp: 
    secs: 1579001767
    nsecs: 970968628
  frame_id: ''
status: 
  - 
    level: 0
    name: "sick_tim310s01: /scan topic status"
    message: "Node starting up"
    hardware_id: ''
    values: []
---
header: 
  seq: 1
  stamp: 
    secs: 1579001768
    nsecs: 980657780
  frame_id: ''
status: 
  - 
    level: 2
    name: "sick_tim310s01: /scan topic status"
    message: "LIBUSB - Read Error."
    hardware_id: ''
    values: []
---
header: 
  seq: 2
  stamp: 
    secs: 1579001768
    nsecs: 980776425
  frame_id: ''
status: 
  - 
    level: 2
    name: "sick_tim310s01: /scan topic status"
    message: "SOPAS - Error reading variable 'DeviceIdent'."
    hardware_id: ''
    values: []
`

mintar, 2020-01-14:
Please see my comment here: https://github.com/uos/sick_tim/issues/35#issuecomment-203858702

Nepomuk153, 2020-01-15:
Sorry for renewed issue. Thought that firmware S1.22 is below V2.51-21.08.15.

Nepomuk153, 2020-01-17:
The technical support approved the Sensortyp TIM310-1030000S01 and we were reciving over usb distance measurment with the terminal in SOPAS. Any other idea about my problem?

mintar, 2020-01-17:
Ok, good to know (especially the fact that you can recieve distance measurements in SOPAS) - so it shouldn't be the firmware version. For reference, we have a TIM310-1030000S01 (which should be the exact same one that you have, right?) with firmware version S1.14-08.05.12, and it works.

My next guess would have been the udev rules, but you already stated that they are installed. Plus, if the permissions were incorrect, it should print the following error, which it doesn't:

```
LIBUSB - Cannot open device (permission denied?); please read sick_tim/README.md
```
```

So it seems that the device is there, the permissions are correctly set by udev, and libusb can connect to the device, but cannot read any data. I'm a bit at a loss here. Could you open the device in SOPAS again and check if you can reset all the settings on the scanner to the defaults? Especially the protocol - it has to be the ASCII protocol (CoLa A), not binary (CoLa B). I'm guessing here, I haven't checked again which settings SOPAS provides for the scanner.

Nepomuk153, 2020-01-17:
Many thanks for your fast reply!
I loaded the basic settings and checked the protocol in the terminal. Seems  to be in (Cola A).
[image]

Still the same error.

Beside the USB connection, i only connected to PIN1 12 V and PIN5 to ground with a power supply device.

mintar, 2020-01-17:
The power supply is working, otherwise you wouldn't be able to access the device via SOPAS. The general form of the telegrams in your screenshot also looks ok; the start (`sSI 18`) looks a bit funny (our scanner sends `sSN LMDscandata` instead), but the rest is identical to what our scanner sends:

```
<STX>sSN LMDscandata 1 1 B96518 0 0 62E7 62E8 653C3EEE 653C4EE6 0 0 8 0 0 5DC 36 0 1 DIST1 3F800000 00000000 FFF92230 2710 10F 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 67B 96 9B A2 AD A3 AA A9 A8 9D 91 EEC EB4 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 324 343 33B 334 33A 339 2B5 23C 239 23A 239 23B 242 244 248 24B 256 256 25B 25F 275 28A 0 379 35F 35C 35C 70A 70D 711 71B 720 720 427 3F6 0 87A 9A5 9E6 B8A B8A 79D 758 6DB 6C1 6D2 6DA 711 6FC 6F2 6F3 70A 720 720 0 0 0 0 0 0 0 90E 82D 7F8 7EA 803 7E9 BEA BBE 6FE B5D B33 B04 AE4 ABF AA1 A7A A5C A3F 9D5 A02 9E7 9CF 9BF 9A6 98F 978 968 957 949 934 922 0 970 8F9 8EE 8E2 8D9 879 8C8 8C0 8B0 8AD 8A7 8A8 8A1 89F 89A 898 893 896 893 893 38C 894 899 89B 8A0 89E 8AB 8AD 8AE 8B8 8C1 345 32E 8DA 8EA 8F3 900 90C 8E5 929 93B 949 95F 96F 981 9ED 0 9C6 9DA 9F7 A0E A27 A4F A69 A4E AAE AD6 AF3 B22 B4C 7F8 80E 807 805 7D9 7D8 7F9 7FF D48 D82 9B8 9AA 0 AB 94 9C 9C 96 9D A3 9B 94 94 94 E76 F1B F4A 0 0 0 0 0 F0B 8D2 8BB 948 9A8 0 0 0 0 0 0 0 0 0 0 F36 0 0 0 0 0 0 0 0 1 RSSI1 3F800000 00000000 FFF92230 2710 10F 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 3A 88 A4 B5 BA AD BB C1 BE 70 1A 2C 36 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 15 75 7B 87 87 86 6D 8E 91 8D 91 95 87 8E 90 93 94 91 91 88 82 69 0 1E 5C 2F 2F 6A 6F 65 63 5F 5F 45 16 0 4 32 4F 54 54 54 55 43 52 48 55 52 68 63 69 63 61 61 0 0 0 0 0 0 0 48 44 63 67 67 2E 53 50 11 54 58 5C 66 5A 5D 60 60 67 1C 60 65 71 71 6C 6D 6E 6B 70 6A 6E 77 0 12 72 76 79 70 36 75 73 76 76 7B 70 6F 70 75 73 73 70 6C 6C 68 6D 73 75 70 76 69 71 71 76 71 6C 67 75 6F 71 6C 64 2E 6B 75 69 6A 6E 6D A 0 63 66 66 62 63 62 5E 25 5E 56 5A 5B 51 D 63 5B 6B 3F 3B 62 49 40 49 1A 31 0 57 93 B3 B1 A6 B0 B2 AE A7 89 89 6 2F 18 0 0 0 0 0 13 5D 5E 33 44 0 0 0 0 0 0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0 0 0 0<ETX>
```

But we're getting ahead of ourselves - we don't have to worry about the content of the telegrams yet, since the driver is not receiving any of them.

By the way, the error code (-7) means `LIBUSB_ERROR_TIMEOUT`, so the driver is trying to read from the USB device and not getting any answer:

https://github.com/libusb/libusb/blob/9eda802d947d9c4212eb3f821fa51956029dade0/libusb/libusb.h#L1077

This whole issue seems to be very similar to #29. If I read that thread right, there was no definitive solution. So I'm fresh out of ideas - maybe try a different computer with a different USB interface, change the USB cable etc.?

mintar, 2020-01-17:
I just found commit 000441dd. Looks like we had to increase the timeout specifically for the TiM310 before; so maybe your should also try to just increase the `USB_TIMEOUT` and see if that helps?

Nepomuk153, 2020-01-20:
Many thanks for your help! 
The default value of the USB_TIMEOUT is already changed to 1000, so i tried 2000 ms and more but without any change.  I also tried to run on another ros distro melodic with no success.
I will check it with another firmware.

mintar, 2021-06-04:
Closing this issue as it was inactive for a long time. Feel free to reopen if you want to add something!

closed 2021-06-04 by mintar