uos/sick_tim #88: SICK TIM320-1131000 - V2.51 --> This is a TiM32x, firmware version 2.51 Failed to init scanner: 2

qwertz747, 2020-02-10:
Hello, I ran into a challenge with the package based on the firmware version 2.51, which is deployed on my lidar (**TIM320**-113).
- On Sopas, connected via USB, the lidar works and the **individual data points are being displayed**.
So it seems that the information regarding the ranges is being transmitted via USB. 
Attached an export from the data received by Sopas. 
[daten.txt](https://github.com/uos/sick_tim/files/4183163/daten.txt)

Unfortunately when I try to use the Ros package I always get the Errors:  
- [ERROR] [1581369680.044696172]: This **scanner model/firmware combination does not support ranging output!**
- [ERROR] [1581369680.044815755]: Supported scanners: TiM5xx: all firmware versions; TiM3xx: firmware versions < V2.50.
- [ERROR] [1581369680.044882317]: This is a TiM32x, firmware version 2.51
- [FATAL] [1581369680.044942473]: Failed to init scanner: 2

Is there a way to get the TIM320 to work? 
Is there a way to reset the firmware version on my lidar to pre V2.5?

qwertz747, 2020-04-29:
Hi,
I got the lidar (TIM320-113) partially to work, but ran in the following error "**received less fields than expected fields** (**actual: 303**, **expected: 580**), ignoring scan".

Attached the output from running " roslaunch sick_tim sick_tim310s01.launch" and the received diagnostics from "rostopic echo /diagnostics". 

I am operating the lidar with 18V, could this be the issue of the less received fields? - it is rated from 9V-28V

Has anyone seen this error before and has a tip how to solve it?

[ INFO] [1588181431.548263846]: SICK AG - TIM3XX - [1]
[ INFO] [1588181431.548773791]: ----------------------------------------
[ INFO] [1588181431.548788322]: Device Class: 0x0
[ INFO] [1588181431.548796551]: VendorID:     0x19a2
[ INFO] [1588181431.548803775]: ProductID:    0x5001
[ INFO] [1588181431.548810606]: ----------------------------------------
[ INFO] [1588181431.548823165]: Interfaces: 1
[ INFO] [1588181431.548831065]: ----------------------------------------
[ INFO] [1588181431.548838981]: Number of alternate settings: 1
[ INFO] [1588181431.548846083]: Interface number: 0
[ INFO] [1588181431.548854960]: Number of endpoints: 2
[ INFO] [1588181431.548863225]: Descriptor Type: 5
[ INFO] [1588181431.548877291]: EP Address: 129
[ INFO] [1588181431.548888457]: Descriptor Type: 5
[ INFO] [1588181431.548898322]: EP Address: 2
[ INFO] [1588181431.548911217]: ----------------------------------------
[ INFO] [1588181431.548957877]: LIBUSB - Claimed interface
[ WARN] [1588181432.086126372]: received less fields than expected fields (**actual: 303**, **expected: 580**), ignoring scan

What got it working so far was the comment #33 

Files:
[diagnostics.txt](https://github.com/uos/sick_tim/files/4553409/diagnostics.txt)
[lidar.txt](https://github.com/uos/sick_tim/files/4553404/lidar.txt)

Thanks!

mintar, 2020-04-29:
That's good news, getting it fully working shouldn't be too hard now! Could you set the `publish_datagram` parameter to `true` and record a rosbag of the `datagram` topic for me?

qwertz747, 2020-04-29:
Hi,

thanks for the fast reply. =)
Is the rosbag-file supposed to look like that?
[2020-04-29-22-08-44.bag.txt](https://github.com/uos/sick_tim/files/4554412/2020-04-29-22-08-44.bag.txt)
[2020-04-29-22-20-49.bag.txt](https://github.com/uos/sick_tim/files/4554418/2020-04-29-22-20-49.bag.txt)

To record the rosbag I did the following steps:
- edit sick_tim310s01.cpp -> set publish_datagram to true
_"nhPriv.param("subscribe_datagram", subscribe_datagram ,true);"_
- make_isolated for me _"./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release_"
- resource for me _"source ~/ros_catkin/install_isolated/setup.bash"_
- start roslaunch _"sick_tim sick_tim310s01.launch"_
- in separate terminal window 
_"rosbag record -a"_ = 1st file
_"rosbag record -a /datagram"_ = 2nd file

mintar, 2020-04-29:
You changed the "subscribe_datagram" parameter, but you need to set the "publish_datagram" parameter. The rosbags are missing the "datagram" topic (you can check for yourself: `rosbag info 2020-04-29-22-08-44.bag`).

qwertz747, 2020-04-29:
Thanks!! I have corrected it.

mintar, 2020-04-29:
Thanks, this looks good! I'll look into that bagfile as soon as I get to it, hopefully tomorrow.

mintar, 2020-05-01:
I've looked into it, and the `sick_tim551_2050001` node is actually able to read and process your datagrams. This is how I tested it:

```bash
roscore
rosparam set use_sim_time true
# now set the parameter `subscribe_datagram` in `sick_tim551_2050001.launch` to `true` (just for testing without a physical device!)
roslaunch sick_tim sick_tim551_2050001.launch
rosbag play --clock 2020-04-29-22-32-48.bag
rostopic echo /scan
```

qwertz747, 2020-05-01:
Hey thanks a lot for the fast reply and your support! =)

I tested it the following way - and it **seems to work great**, although I receive a warning _LIBUSB_ERROR_TIMEOUT:_

-connected the lidar 
-roslaunch sick_tim sick_tim551_2050001.launch
--than I get the following warning: 
[ WARN] [1588348685.428370476]: LIBUSB - Read Error: LIBUSB_ERROR_TIMEOUT.
--I attached the corresponding rosbag [2020-05-01-18-02-45.bag.txt](https://github.com/uos/sick_tim/files/4565179/2020-05-01-18-02-45.bag.txt)
[sicktim_551_terminal.txt](https://github.com/uos/sick_tim/files/4565183/sicktim_551_terminal.txt)
-(in new terminal): rosrun rviz rviz
[image]

So it seems to work just fine. Do I have to do something about the LIBUSB warning?

closed 2020-05-05 by qwertz747