uos/sick_tim #98: TIM510 Scandata missaligned and distorted

maxkremmel, 2022-11-12:
Hello,
I think by looking at the following two images, my problem becomes obvious.
In SOPAS ET the scan looks correct:
[image]
But in rviz the scan is rotated about 45 degrees counterclockwise, and it is also distorted. All contours should be perpendicular to each other:
[image]

Are there some parameters I missed so far to tune the sensor?

Steps to reproduce second image:
- Power Sensor witch 12V
- Connect Sensor via USB witch PC
- execute command "roslaunch sick_tim sick_tim310s01.launch"
- open rviz and add LaserScan and TF

maxkremmel, 2022-11-12:
I found a workaround...

I edited the sick_tim310s01_parser.cpp file to set the following values of the sensor_msgs/LaserScan message published over the topic /scan, manually:
angle_min: -2.356194496154785
angle_max: 2.356194496154785
angle_increment: 0.01745329238474369

Seams like something goes wrong when calculating angle_min, anlge_max and angle_increment.

But I still think I did something wrong because according to #97, the sick_tim310s01 should work without modifications.

mintar, 2022-11-14:
I'm glad you found a workaround! Indeed, it looks like the 3 values you changed manually were not correctly parsed from the datagram (especially angle_increment). I don't have time to debug this any further, but if you want to submit a pull request, I'll be happy to review an merge it.

Also there's an extended fork of this repo here: [sick_scan](https://github.com/SICKAG/sick_scan). Maybe you have more luck with that repo.