uos/sick_tim #99: ROS2 Migration

Noel215, 2023-03-01:
Hi!

Is there any plan to port this repository to ROS2?

mintar, 2023-03-01:
No, this repo is in maintenance mode, and I'm not planning on doing further development on it. There is a more up-to-date fork of this repo here, which also supports ROS2: https://github.com/SICKAG/sick_scan_xd

Noel215, 2023-03-01:
I already opened an issue in the past asking for USB support, but they are not interested in maintaining it so I wanted to know if there was any plan to port/maintain it in this repository for ROS2.

Thanks!

mintar, 2023-03-01:
You mean, `sick_tim` has USB support and `sick_scan` doesn't, right? And you want to ask about porting sick_tim to ROS2?

If you want to prepare a pull request to port sick_tim to ROS2, I'd be happy to review and merge it!

mintar, 2023-05-02:
Yes, thanks for the ping. I just created a new branch `humble`, rebased your PR and merged it. BTW, we have a Tiago robot as well (but still on ROS Kinetic). :)

closed 2023-05-02 by mintar